subsurface/subsurface #136: Line breaks not working when printing in macOS

fezzington, 2017-01-06:
When printing in macOS, line breaks entered in the notes section are not properly shown, and instead show `<br>` tags. 

Screenshots:
Notes section in Subsurface:
[image]
Notes section when printing:
[image]

This is broken in the Flow Layout, One Dive, Six Dives, and Two Dives default templates.

Example file:
[lineBreakTest.xml.zip](https://github.com/Subsurface-divelog/subsurface/files/690719/lineBreakTest.xml.zip)

atdotde, 2017-01-06:
Works for me on my Mac, both for current master as well as the Beta from the Web site.

fezzington, 2017-01-07:
What version of macOS? I'm on 10.11.6, and can't upgrade due to some software I need being broken on newer versions. I wonder if it's just an El Capitan problem?

The same file works for me in an Ubuntu VM using the Beta.

Subsurface-divelog, 2017-01-07:
That should not be related to the OS version. And the Qt version is bundled. Can you give us a screenshot of the About Subsurface screen? Did you compile the binary yourself or is this one of the ones I built?

fezzington, 2017-01-07:
Sure:
[image]

And this was first found on the Beta 1 build, and still shows on the Beta 2 build. I've not tried compiling myself, though I probably could if need be.

Subsurface-divelog, 2017-01-07:
Oh bother. I now have one Mac with 10.11.6 where I can reproduce your problem, and one where I can't. With the exact same binary. This is going to be painful to debug.

Subsurface-divelog, 2017-01-07:
I think I figured it out. Subsurface is picking up an older template that you saved locally at some point. And so the dive.notes aren't marked as 'safe'.
In the print dialog, select the template you want and click on Edit. Now select Template and scroll all the way down to the bottom (and to the right as this is deeply indented). You'll see something like this

```
<div class="textArea">
                       <p> {{ dive.notes }} </p>
</div>
```

but it should read like this:

```
<div class="textArea">
                       <p> {{ dive.notes|safe }} </p>
</div>
```

Add the '|safe', save the change and apply it and it should all work.

fezzington, 2017-01-07:
That fix works, across all of the templates. 

My only worry is that I don't remember ever having printed or viewed that menu before -- and I've only had Subsurface installed since October. Is it possible that it's picking up an old default?

Subsurface-divelog, 2017-01-07:
I honestly don't know. I'll add this to the FAQ

closed 2017-01-07 by Subsurface-divelog