subsurface/subsurface #140: [Idea] Default Startup on View>All

Ghalr, 2017-01-09:
Hello.
I'm a new user of your software and also a fresh driver, and I would like to give an idea.
Would it be possible to open Subsurface directly on View>All ?
Indead i was surprise when I left Subsurface on Globe View to still start on this view when I re-open Subsurface.

Really appreciate,
Best regards.

Subsurface-divelog, 2017-01-10:
Similarly, other users will tell us that they want to come back to the last view that they used. Not something we can make everyone happy with. And Ctrl-1 will fix this for you

closed 2017-01-10 by Subsurface-divelog

Ghalr, 2017-01-10:
Hello.

I do understand.
Maybe an option to check:  Startup using default view (who is View>All) who can also be a Welcome page to start with. ?

Best regards.