subsurface/subsurface #142: [Idea] Add a diving mode

Ghalr, 2017-01-09:
Hello,

Would it be possible to had a Diving Mode (many as pre-saved tags?) to be able to categorize more efficiently our divelogs?

For example:
- Scuba Diving
- Free Diving
- Cave Diving
- Deep Diving
- Technical Diving
- Archaeology‎ Diving
- Biology Diving
- Photography Diving
- Student/Train Diving (for new/experienced divers training for courses)
- Instructor Diving

Really appreciate,
Best regards.

Subsurface-divelog, 2017-01-10:
We have OC/CC/Freediving as diving modes
What you describe is exactly what the tags are intended to do.

closed 2017-01-10 by Subsurface-divelog

Ghalr, 2017-01-10:
Hello.

Yes you are right I was more thinking about auto-completion tags, If we type for example S it should give use Solo, Scuba tags directly. Was not thinking about the custom way.