subsurface/subsurface #144: [Idea] Add Divesite near Location would be better for sorting

Ghalr, 2017-01-09:
Hello

Would it be possible to add a Divesite near the Location already here?
It would be interesting to sort by Location (country) and by Divesite separately, and saw exactly if we already went on that Divesite or not.

Something like:
Location: France, Porquerolles       can list like X divesites
Divesite: Les Mèdes                       can list all diving sessions.

Really appreciate,
Best regards.

Subsurface-divelog, 2017-01-10:
There are many enhancements we should make to the dive site handling and this fits in that list. Unfortunately none of the developers with the UI skills to do this are interested to work on the divesite management (as we found out over the past 18 months). So it's very hard to tell if and when this would be added.

Ghalr, 2017-02-11:
Hi. 
Any news about it ?
Thanks.

dirkhh, 2017-02-11:
As I wrote above... "Unfortunately none of the developers with the UI skills to do this are interested to work on the divesite management (as we found out over the past 18 months). So it's very hard to tell if and when this would be added."
Or to say it more bluntly: yes, that's a great idea. I don't see it happening.

Ghalr, 2017-02-11:
Heyhey Thanks feel free to closed it then, I was not sure with all those commit happening if this happens. :)

dirkhh, 2017-02-11:
I didn't ask you to close it - I'm fine having this good idea around. I just wanted to make it clear that right now no one is working on it. It's possible that at some point a developer with the right skills will be looking for ideas, so having this around (and you available to answer questions if they need more information) might be very helpful.

neolit123, 2017-09-23:
@tcanabrava knows this code the best. maybe he can elaborate if this is a complicated change-request.

tcanabrava, 2017-10-02:
This is mostly not hard. Let me try.

tcanabrava, 2017-10-02:
I think I implemented it on https://github.com/Subsurface-divelog/subsurface/compare/master...tcanabrava:countryOnDivesite

dirkhh, 2017-10-02:
I'm not sure you implement all of what @Ghalr was looking for, but you did implement saving the country as its own item. Now the question becomes "how does one filter dives based on dive sites in one country"?

tcanabrava, 2017-10-02:
he didn't asked to filter, he asked to sort it. I added the country so it would appear on the dive list as a column and he could sort it by country.

dirkhh, 2017-10-02:
ah, my mistake.
Of course, one fun thing we could do... if there is GPS information and the geo-tagging feature gives us a country, we could show that in the column automagically... and only save a different value if the user overrides it with a value that's, err, different.
I'll send you a file to show what I mean.

neolit123, 2017-10-07:
@dirkhh @tcanabrava 

can we close this issue?

dirkhh, 2017-10-07:
I don't think this is exactly what @Ghalr asked for. It gets us closer, though.
Let's wait for @Ghalr to comment and maybe explain more what is still missing (or if he is happy with what we have now)

neolit123, 2017-10-11:
@Ghalr
hi, we might close the issue. do you think the latest Subsurface master gives you the desired results?

neolit123, 2017-11-17:
closing due to inactivity.

closed 2017-11-17 by neolit123