subsurface/subsurface #149: SAC calculation broken

atdotde, 2017-01-10:
For some time, I had the impression that the SAC is computed incorrectly. Now, I came across the dives in 

dives/sac-test.xml

which are test dives that are designed to have a SAC of 20l/min but subsurface consistently finds only 18.5l/min

This will need a bit of bisecting...

glance-, 2017-01-10:
Maybe write a unit test which verifies it?

I also see that weirdly the value increses over the dive to end up with 20l/min. Maybe we're decreesing the pressure by a fixed amount of bar's and when we apply our compression calculations we're offsetting the calculations.

I tested to change the start and end pressure from 202.6bar -> 101.3bar to 101.3bar -> 1bar and the sac is closer to 20l/min, so it indeed looks like a side effect from @torvalds work.

Subsurface-divelog, 2017-01-11:
I believe that Linus showed us that the calculations are likely correct. So we should update the test?

closed 2017-01-11 by Subsurface-divelog