subsurface/subsurface #153: Print formatting notes

lucagrossi85, 2017-01-11:
You can't print notes respecting formatting.
Also if I save and then print my plans immersion, into field notes I see html code.

atdotde, 2017-01-11:
Have a look at the last question of https://subsurface-divelog.org/faq/ This sounds like it could be your problem.

Subsurface-divelog, 2017-01-11:
please report back here if this worked (and close the issue) :)

lucagrossi85, 2017-01-12:
If you select Log > Plain dive from menu bar, then you click button save, after click from menu bar file > print, in this litte window open select one dive with template and click preview, a new little window is open and you can see in the field Notes html code: 

<div><b>DISCLAIMER / WARNING: THIS IS A NEW IMPLEMENTATION OF THE BUHLMANN ALGORITHM AND A DIVE PLANNER...<b></div><br><div><b>Subsurface dive plan</b><br>based on B.....ith GFlow = 30 and GFhigh = 75</div>.

I don't saw my Notes formatted into page print.

dirkhh, 2017-01-16:
did you take a look at the FAQ that @atdotde mentioned above, @lucagrossi85 ?
It seems quite likely that you have a local copy of an outdated print template which is causing this behavior, the faq contains the necessary steps to fix this. If you tried that and it didn't work, please let us know here so we can investigate further.

lucagrossi85, 2017-01-19:
Ok now it work. Thank you all!!

closed 2017-01-19 by lucagrossi85