subsurface/subsurface #158: Application crashes immediately when typing a comma into coordinates

hhmeyer, 2017-01-14:
I use Subsurface 4.5.6 on a Mac with Mac OS X 10.11.6.
When I edit the data for a location and enter the coordinates manually, typing a comma immediately crashes the program. I have set the format of coordinates to decimal.

fezzington, 2017-01-14:
I can confirm this is also happening on 4.6 Beta 2 on 10.11.6.

It does not appear to happen on Beta 2 on Ubuntu 14.04 or Windows 7 though, so it's another macOS thing.

dirkhh, 2017-01-14:
Thanks for the report and the verification that it's still happening in the current source. I'll look into this.

dirkhh, 2017-01-14:
I pushed a new daily build that should fix the bug. Please test it here:

http://subsurface-divelog.org/downloads/daily/Subsurface-4.5.97-45-g1ab6e7fc8a76.dmg

closed 2017-01-14 by dirkhh

fezzington, 2017-01-14:
It's fixed and working with the new daily.

commit 1ab6e7fc (2017-01-14): Prevent crash on Mac when typing comma in GPS coordinates field

Our attempt to skip any white space after the comma causes an ASSERT (strangely
only on Macs).

This closes #158