subsurface/subsurface #165: [Android] Cannot see text at top of screen

cmacdonnacha, 2017-01-18:
Hey,

I'm using the android app and the text at the top of the main screen is truncated. See screenshot.

My phone is on Android  version 6.0.1 and the device is OnePlus Three.

Thanks,
Cathal

[image]

dirkhh, 2017-01-19:
Happy to see your patches. Or the native Android app that you announced in your Google Play review.

closed 2017-01-19 by dirkhh

cmacdonnacha, 2017-01-20:
@dirkhh not sure what you mean by "happy to see your patches"?

dirkhh, 2017-01-20:
That's how open source works. People see something that doesn't work, figure out what causes it, and then submit patches. You do realize that this project is written and maintained by all volunteers, that no one sees a penny for the tens and hundreds and thousands of hours they spend. Quite the contrary, some of us spend a lot of our own money to provide infrastructure (like the cloud storage) so others can use it for free (and then complain about it).

cmacdonnacha, 2017-01-21:
Ah ok now in understand, you'd be happy to see my commits to fix this. Apologies I didn't mean to sound ungrateful and as I said in my review on the play store I hate to leave a poor review but it's an honest review regardless of it being maintained by volunteers or not. You should probably leave this issue open so that it can be tracked.