subsurface/subsurface #166: "Configure dive computer" fails for Oceanic 2.0

GitBobMcG, 2017-01-20:
Able to connect to USB Oceanic Geo 2.0 and download dives from the computer.  Can also connect via the “Configure dive computer” window, but “Retrieve available details” fails.  My Geo 2.0 has version 1I (2014) firmware, which is the most current available.

[image]

glance-, 2017-01-21:
As you see in the left column of that dialog, it supports the OSTC3-family, the Suunto Vyper family and the OSTC family of computers and nothing else.

It certainly doesn't support the "Oceanic Geo 2.0" computer.

dirkhh, 2017-01-21:
Yep, not supported.

closed 2017-01-21 by dirkhh