subsurface/subsurface #168: Handling of multiple buddies in Subsurface-mobile is still broken

dirkhh, 2017-01-22:
a) you can edit the text "Multiple Buddies" and then that becomes your buddy
b) when editing multiple dives one after another, Subsurface-mobile showed me the two buddies from the previous dive after I saved a dive. Switching dives fixed that second problem

There are two ways to go with this:
(1) easy fix - make "Multiple Buddies" read only
(2) real fix - make it truly handle edits to multiple buddies; at least some of the obvious cases - and if that algorithm gets confused, restore the previous buddies (after showing a warning)

dirkhh, 2017-01-23:
So these two commits do (1).
I still wonder if it wouldn't be nicer to have the combo box but also allow the editing of that single line. So if the user taps on the combo box, they get the selection of single buddies. But they can edit in place the comma separated list and when getting a response that contains commas we try to do the right thing and tokenize it. What do you think @jbygdell ?

jbygdell, 2017-01-23:
Initially (1) will be enough as it is more important to not corrupt any data that to be able to handle multiple buddies.

That being said, now that I can build the mobile version again I can look into making (2) work, one issue here is that the last time I fiddled with this the different platforms Android, iOS and Linux behaved differently, for example auto-complete did not work on iOS.

PR #178 (2017-01-29): QML UI: Parse multiple buddies when editing a dive
This allows the user to enter multiple buddies as a comma separated list,
the "Multiple Buddies" entry is still a special case as we can only populate
the combobox with a single name for each entry.

fixes #168
commit 0e570330 (2017-01-29): QML UI: Parse multiple buddies when editing a dive

This allows the user to enter multiple buddies as a comma separated list,
the "Multiple Buddies" entry is still a special case as we can only populate
the combobox with a single name for each entry.

fixes #168
closed 2017-01-29 by dirkhh