subsurface/subsurface #169: Divesite map  all black on Windows 10

gordan0810, 2017-01-23:
Hello,
On Subsurface 4.6.0 divesite map is all black, only dive flags are visible. Subsurface was upgraded from previous release, not clean install.
While on beta Subsurface 4.5.2-1572 map is visible. 
Pictures included.

OS: Windows 10

best regards,
Gordan
[image]
[image]

dirkhh, 2017-01-27:
I cannot reproduce this. Usually an all black map is an issue with accessing the Google server to get map tiles. Do you get an all black rendering for all locations? What happens when you zoom in or out?

gordan0810, 2017-02-01:
Hello,
same is with all locations, and zooming in and out. I even make clean install of Subsurface and the problem is the same.
I tried also to change default browser from Firefox to Edge and Chrome, but problem is the same.
Gordan

dirkhh, 2017-02-01:
That is very strange - I tried a fresh install here on a Windows 10 VM and have no problems with the maps. If you go back to an older version (e.g. 4.5.6), then the maps show up again (you posted a picture from 4.5.2-1572 above - was that after you had upgraded to 4.6 and you downgraded to 4.5.2-1572, or is that just an older screenshot that you happened to have)?

The reason I'm asking is that we "simply" (dangerous word) download the map tiles from a Google API. Now it is possible that there's a bug that we just haven't noticed yet, but with several thousand people on 4.6 already and only one report, I'm not sure that's a likely explanation. Other options could be that you IP was blocked by Google for some reason. Or space aliens. Ok, maybe not that one.

gordan0810, 2017-02-02:
Hello,
both are installed on the same computer and here open in the same time.  I upgraded the general release with 4.6 . They are installed on different places (folders) it the same computer. Here is the picture with the same location.
I can live with it and wait for 4.6.1.
I am not blaming aliens but gremlins.
[image]

dirkhh, 2017-02-02:
I know that this is a frustrating answer for you, but what you show makes no sense to me. Both versions should access the same location for the cached map tiles. And if one finds the tiles, the other should as well. That is assuming they are both run by the same user.
The map data should be in C:\Users\<yourlogin>\AppData\Local\.marble\data\maps\earth\googlesat with a bunch of subdirectories below that which have numeric names.
Nothing in the handling of the map data, in getting it from the server, in storing it locally and in accessing it at runtime has changed from 4.5 to 4.6, and nothing will change in 4.6.1
I am at a complete loss why this is happening for you.
I will create a daily build that creates more information when run from the command line, maybe that will tell us what is going on, but as far as I'm concerned I have not the slightest idea where to even start looking for this problem.

dirkhh, 2017-02-02:
OK, here's an attempt to collect more data. Please download and install this Windows installer
http://subsurface-divelog.org/downloads/daily/subsurface-4.6.0-41-marble-debug-g8e48c6a06820.exe
open a cmd.exe (for example by pressing Windows-R and then entering cmd.exe) and run
`"C:\Program Files (x86)\Subsurface-debug\subsurface.exe" --win32log -v -v
`
(or whatever path you used to install that version of Subsurface).
Open your dive log file, verify that you see a black map, quit Subsurface.
In your home directory (or more specifically, the current working directory of your cmd.exe when you started Subsurface) you will find subsurface_err.log. Please attach this log file to this issue - maybe that will give us enough useful information to figure out what's going wrong.

gordan0810, 2017-02-03:
Hello,

first of all, I checked the path, and I do not have the folder you mentioned.
I have only info about cloud storage, which I cannot access at the moment. It is in Roaming/Subsurface folder. Only folders that I have in AppData are Local, LocalLow and Roaming.
In Local is gtk-2.0 folder and in Roaming Subsurface folder.
Anyhow here are err and out log.
They extensions are renamed to txt because of github rules.

Gordan

[subsurface_err.txt](https://github.com/Subsurface-divelog/subsurface/files/750542/subsurface_err.txt)
[subsurface_out.txt](https://github.com/Subsurface-divelog/subsurface/files/750541/subsurface_out.txt)

dirkhh, 2017-02-03:
Subsurface is looking for the map tiles in 
`C:/Users/Gordan/AppData/Local/.marble
`
and cannot load any of them. It seems to me that there is an issue writing to that location since Subsurface believes that it successfully download tiles, but then cannot access them on disk.

What's the output of
`cd C:\Users\Gordan\AppData\Local
dir /a
cd .marble
dir /a
`

gordan0810, 2017-02-06:
Here
[dir_local.txt](https://github.com/Subsurface-divelog/subsurface/files/754634/dir_local.txt)
[dir_marble.txt](https://github.com/Subsurface-divelog/subsurface/files/754636/dir_marble.txt)

Also I checked data folder and there are google satellite pictures of dive site locations
[dir_googlesat.txt](https://github.com/Subsurface-divelog/subsurface/files/754639/dir_googlesat.txt)
but there is also folder streets and all pictures in subfolders are black (at least those that I sow). If I rename it, opening of Subsurface recreates this folder and result is the same.
In the same time old beta is OK.

Gordfan

dirkhh, 2017-02-07:
The thing that I don't understand:
a) why would those pictures in the subfolders be black?
b) how can the beta show correct maps - it access THE SAME FOLDERS

gordan0810, 2017-02-17:
Is there any possibility to completely remove Subsurface, and start installation from scratch?
I mean everything, preferences, registry entries.
I have cloud logbook with Subsurface so it does not matter.

Gordan

dirkhh, 2017-02-17:
Subsurface comes with an uninstaller (it's in the install directory that you picked at install time) - but IIRC we decided against removing the data cache directories from the user's computer.

dirkhh, 2017-03-06:
Have you been able to solve this? You are still the only one with the problem (or I should say, none of our other several thousand Windows users has reported it)...

gordan0810, 2017-03-06:
No,
Even the upgrade, as you prophesied, did not change it.
I hope that this can be solved with completely removal of Subsurface. Is there any old installation which include removal of everything related to Subsurface, Or some manual instructions how to do it.
 I do not wan to reinstall windows to remove Subsurface.
Also it may help, I notice that I cannot change the size of the font. Whenever I put another size, with reopening it goes back to 8.25.

dirkhh, 2017-03-06:
As I mentioned in an earlier comment - each version of Subsurface that you install (in different directories) come with an uninstaller.

dirkhh, 2017-04-22:
Nothing in six week, either @gordan0810 has given up (which would be disappointing), or this is fixed. Either way, without being able to reproduce it / understand what's going on, I don't know what to do, anyway.
Closing for now

closed 2017-04-22 by dirkhh

Ghalr, 2017-04-25:
I don't know if this is the bug you guys reported here
Using Windows 8.1, after Zoom-out until I see the Earth globe
http://prntscr.com/f0mo01

Load your divelog in View>All
View>Globle  should be zoom-in
Zoom-out until you see the Earth in Space
Once it's all black, Zoom-In and it's like you are inside the Earth

dirkhh, 2017-04-25:
Oh, never thought about this possible problem - but your right, that could be the problem...

gordan0810, 2017-04-25:
Hello,
I did not give up, but faced a wall.
I installed all possible version from subsurface site, and problem start with first 4.6 release. So all 4.5 are OK, 4.6 none. I always did full deinstalation with Revo Uninstaller but did not help.
Also installed on the another same laptop HP Elitebook 8560w with win 10 and everything was OK. Maps were visible.
Find on the net that there could be a problem with black google satellite maps (it is reported) and then played with 3D in all of my installed browsers. Nothing helps.
Recently I had replaced graphic card in my Elitebook, so my suspicion is that there is a problem. But dxdiag finds nothing.
As I have both Linux Mint Laptop and Mac Pro Book, I decided to use one of them as main Subsurface logbook. And wait until I replace my Elitebook with some other. 

All of this took me a lot of time as I travel, so did not have anything to report.

Ghalr, 2017-04-25:
Also to be honest I do not like the map used.
I mean, this is not clear/hd version

Something like https://www.openstreetmap.org/ would be a good solution.

- ocean are crystal clear (no need to see Weather, or Ocean details)
- street/landscape are also crystal clear (no need to see every detailed, a gps do better if we need to dive into a lake)

Everytime I look into the map, right now, it's like at 5meters distances we do not have a global view.

dirkhh, 2017-04-25:
Hi @gordan0810 - we shouldn't be seeing the Black Google Sat Images as one of my tests here prior to a release is a fresh install in a fresh VM and that would be affected by that problem.

Why 4.6 builds would fail, but 4.5 builds would work on the same laptop continues to be completely incomprehensible to me :-(

@Ghalr you may be in the minority with your opinion. When we first added maps we did indeed use openstreetmap. Our users hated that because many of them were diving in areas where OSM had no viable maps. And we got overwhelming requests to provide Satellite images and not street maps (because these "maps" are about diving, not about driving).

As always - we cannot give every user exactly what they want without making the software completely untestable and unmaintainable and unusable. You hear this from me on a fair share of your proposals - adding more options, more alternatives, comes at a very serious cost. It makes the UI more complicated and makes it harder to test the various combinations. So I don't think we'll add an option to switch to OSM.

Ghalr, 2017-04-25:
I do understand.
I have not checked but do there are an option to export GPS coordinate to a texte file or cvs file ?

dirkhh, 2017-04-25:
the GPS coordinates are in the XML file in plain text. We don't offer a way to just export the coordinates (since most people need the opposite direction - they have the coordinates and want to add them to Subsurface.
It would be easy to create a simple script that takes an exported XML file and prints out the GPS locations. The format looks like this:

```
<site uuid=' fd9bca7' name='Paso del Cedral' gps='20.375400 -87.030600'>
</site>
<site uuid=' ffb1391' name='Blue Corner' gps='7.133800 134.221393'>
</site>
<site uuid='12699f7f' name='Ilha do Meio, Fernando de Noronha, Brazil'>
</site>
<site uuid='12c7055a' name='Amazing Maze' gps='-17.248400 178.194000'>
</site>
```

gordan0810, 2017-04-25:
Hello,
I checked with virtualbox with MSEdge machine from MS. Installed Suburface 4.6.3 and images are visible. I suspect about problem in my graphics card.