subsurface/subsurface #170: Changed Dive No. not saved...

tinoo, 2017-01-23:
Hi Folks
I was just updating some dives with additional information, when i discovered the following behavior:

If i double click on the dive number at the Dive List and change the dive number, this will not be saved (into cloud)! Next time when i open subsurface again the numbers are like before...

The new dive number will only be stored when i right click -> renumber dive(s) and change the number there.
If you want to change more than one dive, this becomes very annoying... Double clicking directly into the Dive List would be much faster...
Maybe someone can put an eye on that issue? Thanks a lot!

jbygdell, 2017-01-23:
You can renumber a continuous range of dives by selecting them, right click -> renumber dive(s), input the number that the first dive in the selection should have. The other dives will then be numbered in ascending order.

dirkhh, 2017-01-27:
We need to figure out how to disable the double click on the dive number

PR #173 (2017-01-28): Ignore double clicks on the divelist
Fixes #170 on GitHub

Simple catch function for double click events in the divelist,
prevents users from trying to edit the divenumber ithe wrong way.
commit b0ec7146 (2017-01-28): Ignore double clicks on the divelist

Fixes #170 on GitHub

Simple catch function for double click events in the divelist,
prevents users from trying to edit the divenumber ithe wrong way.
closed 2017-01-28 by dirkhh