subsurface/subsurface #172: Version 4.6.0 crashes exporting to divelogs.de

grosengarten, 2017-01-27:
Subsurface 4.6.0 crashes when exporting a selection of dives to divelogs.de.  This bug seems to have been introduced by this version- on the previous version exporting to divelogs.de worked fine.

I'm running OS X Yosemite and the logs were imported from a Shearwater Perdix.

dirkhh, 2017-01-27:
I tried to reproduce this and for me it works fine (admittedly, tested on El Capitan as I don't have a access to a Mac running Yosemite right now). Can you send me in email an XML export of your dive data file so I can try to reproduce the problem here?

fezzington, 2017-01-27:
I tried last night and was able to reproduce on El Capitan, using the official binary download. Was also crashing when trying to export all. 

I've never exported to divelogs before, if that makes a difference

dirkhh, 2017-01-27:
Can you send me the crash report (as produced by MacOS) - it might help a little. Or send me your XML file... I'd love to figure out what's going wrong here

dirkhh, 2017-01-28:
I believe I identified the build problem that caused this. The libzip-1.1.2 built from current Homebrew apparently doesn't work correctly under certain circumstances - I now built the library manually and packaged that replacement .dylib with an otherwise unchanged Subsurface 4.6. I have replaced the Subsurface-4.6.0.dmg on our download page with an updated version and using that @fezzington was no longer able to reproduce the crash (I was never able to reproduce it). 
@grosengarten would you please test that this new DMG works for you as well and then close this issue?
Thanks

grosengarten, 2017-02-01:
Sorry, where can I find the DMG?

fezzington, 2017-02-01:
@grosengarten , the new DMG can be found on the Subsurface Website, on the [downloads page](https://subsurface-divelog.org/download/) .

grosengarten, 2017-02-01:
Yes, thanks-  found it.  Indeed the issue looks to have been fixed.  Thanks!

closed 2017-02-01 by grosengarten

dirkhh, 2017-02-01:
Thank you so much for reporting back - really appreciated. Let us know if you run into other issues :-)