subsurface/subsurface #174: Visual dive editor rescaling prevents certain dive durations

dirkhh, 2017-01-29:
64 minutes appears to be one of the ones you can't get to:

http://www.mgnewman.com/videos/subsurface.mp4

dirkhh, 2017-01-29:
@atdotde please check that this didn't break anything in the planner...

closed 2017-01-29 by dirkhh

atdotde, 2017-01-30:
> On 29.01.2017, at 06:05, Dirk Hohndel <notifications@github.com> wrote:
> 
> @atdotde <https://github.com/atdotde> please check that this didn't break anything in the planner...
> 
> 

Cannot see why it would. Looks fine to me.

Best
Robert
--
.oOo.oOo.oOo.oOo.oOo.oOo.oOo.oOo.oOo.oOo.oOo.oOo.oOo.oOo.oOo.oOo.oOo.oOo.oOo.oO
Robert C. Helling     Elite Master Course Theoretical and Mathematical Physics
                      Scientific Coordinator
                      Ludwig Maximilians Universitaet Muenchen, Dept. Physik
                      Phone: +49 89 2180-4523  Theresienstr. 39, rm. B339
                      http://www.atdotde.de

Enhance your privacy, use cryptography! My PGP keys have fingerprints
A9D1 A01D 13A5 31FA 6515  BB44 0820 367C 36BC 0C1D    and
DCED 37B6 251C 7861 270D  5613 95C7 9D32 9A8D 9B8F