subsurface/subsurface #176: Include support for Aqualung i750

maxdraki, 2017-01-29:
I believe support is included in more recent libdivecomputer builds. Can this please be added to subsurface

thank you

dirkhh, 2017-02-02:
We pulled the latest libdivecomputer and daily builds are available at https://subsurface-divelog.org/downloads/daily - this should bring in support for the i750 (among several other new models)

Please give this a try and let us know!

maxdraki, 2017-02-05:
Hi Dirk,
Unfortunately I'm getting a seg fault with the daily build on Linux 4.8.0-37-generic #39-Ubuntu

dirkhh, 2017-02-05:
Which Linux distribution? Which build are you using? The AppImage? 

https://github.com/Subsurface-divelog/subsurface/issues/176#issuecomment-277533710

maxdraki, 2017-02-05:
I'm on Ubuntu 16.10 and tried with both the AppImage and the ppa

dirkhh, 2017-02-05:
With the Subsurface from the PPA, would you please run it under the debugger?

gdb /usr/bin/subsurface

run

once it crashes

bt

and paste the output here?

Thanks

maxdraki, 2017-02-05:
Here you go Dirk
`(gdb) bt
#0  0x0000555555804d2d in get_gas_used ()
#1  0x0000555555736e61 in MainTab::updateDiveInfo(bool) ()
#2  0x00005555556753c5 in MainWindow::current_dive_changed(int) ()
#3  0x00007ffff2675b49 in QMetaObject::activate(QObject*, int, int, void**) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#4  0x00005555556f966e in DiveListView::currentDiveChanged(int) ()
#5  0x000055555570cd54 in DiveListView::selectionChanged(QItemSelection const&, QItemSelection const&) ()
#6  0x00007ffff585e881 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#7  0x00007ffff2675b49 in QMetaObject::activate(QObject*, int, int, void**) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#8  0x00007ffff25f6ba7 in QItemSelectionModel::selectionChanged(QItemSelection const&, QItemSelection const&) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#9  0x00007ffff25fb2eb in QItemSelectionModel::emitSelectionChanged(QItemSelection const&, QItemSelection const&) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#10 0x00007ffff25fecbc in QItemSelectionModel::select(QItemSelection const&, QFlags<QItemSelectionModel::SelectionFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#11 0x00007ffff25f883c in QItemSelectionModel::select(QModelIndex const&, QFlags<QItemSelectionModel::SelectionFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#12 0x00007ffff25f6ff4 in QItemSelectionModel::setCurrentIndex(QModelIndex const&, QFlags<QItemSelectionModel::SelectionFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#13 0x00007ffff58580e9 in QAbstractItemView::setCurrentIndex(QModelIndex const&) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#14 0x000055555570a1b9 in DiveListView::reload(DiveTripModel::Layout, bool) ()
---Type <return> to continue, or q <return> to quit---
`

dirkhh, 2017-02-05:
That is code that recently changed. I don't see a crash here, so this may be due to your default dive file... if you use 4.6.0 and start it, could you save that first dive that is shown when you start and send it to me in email (or attach it here if you don't mind)

maxdraki, 2017-02-06:
I've downgraded to 4.6.0 and it works fine and have emailed you the first dive shown (the most recent one). This was imported from my i750 using a a version of libdc that Jef sent me

atdotde, 2017-02-06:
Would you mind mailing the dive to me as well as I am likely the culprit who changed that code recently? helling@atdotde.de

atdotde, 2017-02-06:
I guess that is something I introduced but it should already have been fixed in

atdotde, 2017-02-06:
I guess this is something that was already fixed on Friday in commit 

855995191ca997ddc41e3ce23422e4adbad16949
Author: Robert C. Helling <helling@atdotde.de>
Date:   Fri Feb 3 22:50:10 2017 +0100

    Don't do strcmp on NIL pointer
    
    This is what we have same_string() for...

that is in 	
Subsurface-4.6.0.53-...

At least, it does not crash for me anymore.

dirkhh, 2017-02-06:
I have kicked off new daily builds for version 4.6.0.73 - based on what Robert says, I assume this will fix your problem. Could you check (might be a few more minutes before those are up on the server) and if it does indeed works, would you close this issue?

maxdraki, 2017-02-06:
Thanks both - that seems to work :)

One thing - when using the app image, I launched the app (not integrated into system) and then plugged my DC by USB cable, but the app didn't recognised /dev/ttyUSB0 until I exited the app and restarted it.. Not sure if that's an expected result of the appimage or not

closed 2017-02-07 by maxdraki