subsurface/subsurface #177: Windows XP

digidiver11, 2017-01-29:
Hello everybody,
I've been using the software "Subsurface" for quite some time now. At this point, a hearty thanks to the developers!
Unfortunately, I have the problem that version 4.6.0 no longer works under Windows XP.

Please do not teach about Windows XP. I know how and why I use this operating system.
The topic Windows ends with me anyway with Windows 7, afterwards there is only still Linux.

Yours sincerely, from Berlin

dirkhh, 2017-01-29:
Not an intentional decision by anyone. Can you give me more detail on what happens when you try to start it? I no longer have access to a Windows XP system, so this isn't something I can test / reproduce

digidiver11, 2017-01-29:
Hello Dirk,
    
    If I start the "Subsurface 4.6.0" under Windows XP the following
    error message occurs: The procedure entry point "qsort_s" was not
    found in the DLL msvcrt.dll.
    
Mit freundlichen Grüßen / With kind regards,
 

Detlef Kochan

---------------------------------------
kms.kochan@gmx.net
Tel.: +49 172 / 391 61 79

*******************************************************************************************
Diese Nachricht ist vertraulich und ausschließlich für den Adressaten
bestimmt. Jeder Gebrauch durch Dritte ist verboten. Falls Sie die Daten
irrtümlich erhalten haben, nehmen Sie bitte Kontakt mit dem Absender
auf und löschen Sie die Daten auf jedem Computer und Datenträger.

Es ist ausdruecklich untersagt, die E-Mailadressen dieser E-Mail zur 
Weitergabe an Dritte, Werbezwecke, Statistiken, o.a. zu verwenden !!!!!
Jede Zuwiderhandlung wird strafrechtlich verfolgt !!!!!!

This message is confidential and intended solely for the use by the
addressee. Any use of this message by a third party is prohibited. If
you received this message in error, please contact the sender and
delete the data from any computer and data carrier.

It is expressly forbidden, e-mail addresses of this e-mail
disclosure to third parties, promotional purposes, statistics, etc. 
to use !!!!!  Any violation will be prosecuted !!!!!!
*******************************************************************************************
    Am 29.01.2017 um 17:20 schrieb Dirk
      Hohndel:
    
    
      Not an intentional decision by anyone. Can you give me more
        detail on what happens when you try to start it? I no longer
        have access to a Windows XP system, so this isn't something I
        can test / reproduce
      —
          it on GitHub, or mute
          the thread.

digidiver11, 2017-01-29:
Hello Dirk,

i have found this here: http://www.tripleboot.org/?p=423

---------------------------------------------------------------------------------------------------------------
Let’s take them in order:

    Open the .pro file for your project and insert this line (I usually do it at the end):

    QMAKE_LFLAGS_WINDOWS = /SUBSYSTEM:WINDOWS,5.01

    Why 5.01? Well if you look at the original settings, using for example dumpbin/headers on your .exe file, you see that the SUBSYSTEM:WINDOWS version is set to 6. This means it can run on Windows Vista and later and why Windows XP rejects it. By changing it to 5, we declare XP and Win2k3 as kosher chaps too. (Those numbers 5 and 6 are the internal Windows version numbers.)
    I thought originally it should be enough to specify /SUBSYSTEM:WINDOWS,5 but then the linker complains: LNK4010: invalid subsystem version number, and it resets it back to 6 🙁 So 5.01 it is.

    Note: if you’re compiling a 64-bits flavored .exe file (and thus targeting 64-bits Windows XP) you instead should set <VALUE> to /SUBSYSTEM:WINDOWS,5.02 

-------------------------------------------------------------------------------------------------------------------------------------

I hope this can help you. Thank you for your help !

With the best regards

dirkhh, 2017-01-29:
Thanks for the input, but we don't build the Windows binaries on Windows but we cross-build them under Linux. Additionally, we don't use qmake.
I'll see if I can find something similar to this for the MXE toolchain, but I will be blunt and honest with you... adding support for a 16 year old operating system is not high on my list of priorities. If it's something simple and easy to do, sure. Otherwise I'm afraid you'll be stuck on whatever is the last version of Subsurface that worked on XP.

digidiver11, 2017-01-30:
Hello Dirk,
    
    Yes, I can
        understand that! Can you explain to me
        briefly how the program for Windows is compiled. Can
        I do so easily?
      Since I do not develop programs for the PC, neither
        for Windows nor for other operating systems, I am here something
        on the hose.
      What do I need. I am a VM
        (Virtualbox), a lean Linux Distri and the appropriate tools.
      I can not do anything with the more elaborate
        explanations on the Internet.
        
        I hope, that you can understand my bad english, sorry. It ist
        from google translator ;-)).
      
    
With kind regards,
 

digidiver11

*******************************************************************************************
    Am 29.01.2017 um 17:50 schrieb Dirk
      Hohndel:
    
    Thanks for the input, but we don't build the Windows
      binaries on Windows but we cross-build them under Linux.
      Additionally, we don't use qmake.
      I'll see if I can find something similar to this for the MXE
      toolchain, but I will be blunt and honest with you... adding
      support for a 16 year old operating system is not high on my list
      of priorities. If it's something simple and easy to do, sure.
      Otherwise I'm afraid you'll be stuck on whatever is the last
      version of Subsurface that worked on XP.

dirkhh, 2017-01-30:
Ich verstehe nicht, warum Du das selber kompilieren willst - davon laeuft das ganze dann auch nicht unter XP. Der Prozess ist relativ gut dokumentiert, ist aber sicherlich nichts fuer den Anfaenger der sich mit Linux nicht auskennt. Und auf einem aelteren PC (ich nehme an, der ist aelter, da er unter XP laeuft) mit einer Linux VM wuerde der komplette Prozess wahrscheinlich mehrere Tage dauern.
Es waere wahrscheinlich viel einfacher, eine VM mit einer Ubuntu 16.04 zu installieren und Subsurface in der VM laufen zu lassen :-)

digidiver11, 2017-01-30:
Hallo Dirk,
    
    ich glaube ich muss da etwas aufklären. Also unterwegs auf Reisen
    habe ich immer ein Netbook verwendet. Diese Lenovo-Netbooks laufen
    unter Win7 extrem bescheiden.
    Mein PC zu Hause sieht schon aus beruflichen Gründen ganz anders
    aus. Dort ist ein i7-3930k-3,2 GHz mit 64 GB RAM im Einsatz.
    Als Betriebssystem läuft meist ein speziell angepasstes Win7Pro x64.
    Auf diesem OS ist dann die Virtualbox installiert. Es laufen mehrere
    verschieden OS auf diesem System.
    
    Meine Frage bezog sich also nur auf mein "Urlaubs-Netbook".
    
    Du hast natürlich vollkommen recht, es lohnt sich einfach nicht mehr
    an dieser alten Krücke festzuhalten. Daher habe ich beschlossen, die
    alten Kisten endgültig bei Ebay zu verticken.
    Im Anschluss kommt dann ein HP Elitebook 2760p zum Einsatz. Wichtig
    ist mir bei diesem Gerät, die Möglichkeit zu haben, verschiedene
    Schnittstellen per Express Card anschliessen zu können. Genau
    deshalb hatte ich noch an den alten Lenovo-Teilen festgehalten.
    
    
    Also ganz herzlichen Dank für Deine Infos. Aber Du hast schön völlig
    recht - neuere Hardware und endlich den kompletten Umstieg auf Linux
    durchziehen.
    
    Nur eine letzte Frage, welches Linux würdest Du mir empfehlen ????
    
Mit freundlichen Grüßen aus Berlin
 

Detlef Kochan

    Am 30.01.2017 um 16:39 schrieb Dirk
      Hohndel:
    
    
      Ich verstehe nicht, warum Du das selber kompilieren willst -
        davon laeuft das ganze dann auch nicht unter XP. Der Prozess ist
        relativ gut dokumentiert, ist aber sicherlich nichts fuer den
        Anfaenger der sich mit Linux nicht auskennt. Und auf einem
        aelteren PC (ich nehme an, der ist aelter, da er unter XP
        laeuft) mit einer Linux VM wuerde der komplette Prozess
        wahrscheinlich mehrere Tage dauern.
        Es waere wahrscheinlich viel einfacher, eine VM mit einer Ubuntu
        16.04 zu installieren und Subsurface in der VM laufen zu lassen
        :-)
      —
          it on GitHub, or mute
          the thread.

dirkhh, 2017-01-31:
Hallo Detlef,

Ich glaube nicht, dass ich explizit gesagt habe, dass es sich nicht lohnt, an Deinem Netbook festzuhalten. Lustigerweise war ich bei Intel beschäftigt, als die Netbooks "erfunden" wurden, und ich fand die von Anfang an eher frustrierent unter unter-powert. Ich denke aber, dass Deine Entscheidung Sinn macht.
Wenn es um Linux-Versionen geht... das ist alles Geschmackssache. Ich denke, dass die beste Unterstützung für Deutsche Benutzer sicher in einer OpenSUSE zu finden ist (full disclosure: ich war vor vielen, vielen Jahren mal der CTO der SUSE). Viele andere schwören auf Ubuntu.
Ich selber benutze Arch Linux, aber das ist eher was für Leute, die bereit sind, sich ein bisschen mit einer komplizierteren Installation und etwas weniger "hand-holding" herumzuschlagen. Ausserdem habe ich keine Ahnung, wie gut die Unterstützung für deutschsprachige Benutzer ist (ich habe in gut 25 Jahren mit Linux noch nie eine lokalisierte Version benutzt, kann da also nicht mitreden - und mein Deutsch wir offensichtlich nach 15 Jahren in den USA auch eher etwas holprig).

Aber angesichts Deiner Antwort werde ich den "issue" jetzt mal schliessen (und die Dokumentation dahingehend updaten, dass XP anscheinend nicht mehr unterstützt ist. 4.5.6 ging noch, richtig?)

closed 2017-01-31 by dirkhh

atdotde, 2017-01-31:
Hallo,

> On 30.01.2017, at 17:04, digidiver11 <notifications@github.com> wrote:
> 
> ich glaube ich muss da etwas aufklären. Also unterwegs auf Reisen
> habe ich immer ein Netbook verwendet. Diese Lenovo-Netbooks laufen
> unter Win7 extrem bescheiden.

ween ich mich da kurz nochmal einschalten darf: Nach meiner Erfahrung laufen Netbooks unter Linux noch halbwegs flüssig, wenn sie unter Windows schon eher wie Honig laufen (aber ja, ein alter Atom ist aus heutiger Sicht schon etwas untermotorisiert). Bevor Du ihn wegwirfst, kannst Du ja nochmal versuchen, ein Linux draufzuhelfen (wenn das mit dem, was Du sonst damit tust, zusammen passt).

Ich habe auch noch ein altes Netbook rumliegen (und musst zu Weihnachten lernen, dass sogar mein Vater sein etwa 10 Jahre altes noch benutzt und ein Ubuntu Update brauchte) und finde, dass sich Subsurface auf den extrem schmalen Displays eher grenzwertig benutzen lässt.

Aus der Distributionsdiskussion halte ich mich raus, aber eines ist wichtig: emacs rulez over vi!!1!

Viele Gruesse
Robert
--
.oOo.oOo.oOo.oOo.oOo.oOo.oOo.oOo.oOo.oOo.oOo.oOo.oOo.oOo.oOo.oOo.oOo.oOo.oOo.oO
Robert C. Helling     Elite Master Course Theoretical and Mathematical Physics
                      Scientific Coordinator
                      Ludwig Maximilians Universitaet Muenchen, Dept. Physik
                      Phone: +49 89 2180-4523  Theresienstr. 39, rm. B339
                      http://www.atdotde.de

Enhance your privacy, use cryptography! My PGP keys have fingerprints
A9D1 A01D 13A5 31FA 6515  BB44 0820 367C 36BC 0C1D    and
DCED 37B6 251C 7861 270D  5613 95C7 9D32 9A8D 9B8F

digidiver11, 2017-01-31:
Hallo Dirk,
    
    und alle anderen, danke für Eure Antworten.
    
    Am besten war der Hinweis zum Thema Notebook mit dem sehr schmalen
    Bildschirm.
    
    Wir machen es ganz einfach, ich begrabe das Thema Netbook (oder
    Frustbook)  und das Thema WinXP
    und Du, lieber Dirk, kannst diesen Thread schliessen.
    
    So ist das Thema endlich erledigt.
    
With kind regards,
 

Detlef Kochan

    Am 31.01.2017 um 07:20 schrieb Dirk
      Hohndel:
    
    
      Hallo Detlef,
      Ich glaube nicht, dass ich explizit gesagt habe, dass es sich
        nicht lohnt, an Deinem Netbook festzuhalten. Lustigerweise war
        ich bei Intel beschäftigt, als die Netbooks "erfunden" wurden,
        und ich fand die von Anfang an eher frustrierent unter
        unter-powert. Ich denke aber, dass Deine Entscheidung Sinn
        macht.
        Wenn es um Linux-Versionen geht... das ist alles
        Geschmackssache. Ich denke, dass die beste Unterstützung für
        Deutsche Benutzer sicher in einer OpenSUSE zu finden ist (full
        disclosure: ich war vor vielen, vielen Jahren mal der CTO der
        SUSE). Viele andere schwören auf Ubuntu.
        Ich selber benutze Arch Linux, aber das ist eher was für Leute,
        die bereit sind, sich ein bisschen mit einer komplizierteren
        Installation und etwas weniger "hand-holding" herumzuschlagen.
        Ausserdem habe ich keine Ahnung, wie gut die Unterstützung für
        deutschsprachige Benutzer ist (ich habe in gut 25 Jahren mit
        Linux noch nie eine lokalisierte Version benutzt, kann da also
        nicht mitreden - und mein Deutsch wir offensichtlich nach 15
        Jahren in den USA auch eher etwas holprig).
      Aber angesichts Deiner Antwort werde ich den "issue" jetzt mal
        schliessen (und die Dokumentation dahingehend updaten, dass XP
        anscheinend nicht mehr unterstützt ist. 4.5.6 ging noch,
        richtig?)
      —
          it on GitHub, or mute
          the thread.