subsurface/subsurface #184: Strange effect in planner: Repetitive dives and surface time

sfuchs79, 2017-01-31:
Found a strage effect in the planner linked to repetitive dives and the new feature which is showing the surface time between two repetitive dives:

Attached file has two planned dives: 1001 (1. Feb 2017) and 1002 (6. Feb 2017)

- In dive 1001 you can already see in the notes that there is "Oberflächenpause 0:06" (surface time) which is obviously not correct.
- If you "Edit dive in planner" for 1001 and remove all the planned waypoints (Ctrl+leftklick on waypoint 6) this wrong "Surface time 0:06" immediately shows up again.
- Ignore this and save the dive again "Save"
- If you "edit dive in planner" for 1002 there is also s.th. strange.
- Again Ctrl+leftklick on waypoint 5 --> Again the issue with "Surface time 0:06" plus runtime is 63min
- Change the date of the dive 1002 to 2. Feb 2017 you see a correct surface time of 22:30 but the runtime now is 61min. Why shorter deco although this now IS a repetitive dive?

I tested all this with current version from git (subsurface-4.6.0-36) because the availabilty of ctrl-leftclick feature and the bugfix for the zero-minute legs make testing this easier.
[Stefan_Planung_Test.zip](https://github.com/Subsurface-divelog/subsurface/files/742908/Stefan_Planung_Test.zip)

atdotde, 2017-02-01:
I cannot reproduce this with current master, which version are you using?

Also note that the notes reflect the dive as it was planned: When you planned it as a repetitive dive (i.e. there is a dive in the dive list in the 48 hours before the start of the planned dive) the stops are calculated taking that into account and the statement about the surface interval stays there even when you delete the previous dives from the dive list.

So please check what your dive list contains when you do this planning. If you just open the xml that you attached, there should be no surface interval (and for me there isn't).

Let me also point out that dive 1001 is planned with slightly unconventional gradient factors 55/45, i.e. GF_low being bigger than GF_high.

If the problem persists even when you just have the two dives in the dive list, it might be due to some uninitialised memory, something that is a pain to track down.

sfuchs79, 2017-02-01:
I did some more tests, here is additinal info:
- Version: 4.6.0-36-g49cf4933df4a
- OS: Now even tested both, linux and win 10
- Found even simpler setup to reproduce:
     o Start subsurface, new logbook
     o Log -> Plan dive
     o I immediately get what is shown on the screenshots
     o Interesting: Different insane surface time values for linux and win 10

Files attached...
[issue_surface_time.zip](https://github.com/Subsurface-divelog/subsurface/files/745698/issue_surface_time.zip)

atdotde, 2017-02-01:
Found it (I needed to install a virtual machine with Ubuntu and build it there). Indeed, it was a memory corruption (or better: uninitialised value). Patch on the way. Thanks for reporting!

PR #185 (2017-02-01): Initialize the surface time
Otherwise we return garbage when there is no dive in the dive list.

Closes #184
commit bb86d135 (2017-02-01): Initialize the surface time

Otherwise we return garbage when there is no dive in the dive list.

Closes #184
closed 2017-02-01 by dirkhh