subsurface/subsurface #188: [Idea] Rock bottom/Minimum gas calculation

sfuchs79, 2017-02-03:
Hi All,

after already starting the discussion with Robert and others on the mailing list I put this as a reminder also here. Meanwhile I plan even to try to do this on my own with maybe the help of a friend of mine who is a SW developer and diver :-)

Either only for my private version or if you finally like it also for the official version.
Feedback is welcome!

What we already have today:
We have "Bottom SAC" GUI user input, cylinder size of last cylinder used in "dive planner points" and we already have tank pressure consumption from this cylinder in planned ascent including all calculated deco stops until we reach first deco gas switch. That's a good starting point.

What we would need for the rock bottom calculation:
- Problem_solving_time (including some margin) as GUI input, integer number of minutes, default e.g. 4min
  This is the worst case time the buddy team will stay at max. depth after OOG situation happened.
- SAC_factor
  This is the increase factor for bottom SAC rate in OOG situation. I would multiply bottom SAC by 2 already hard coded and then again multiply by this factor.

Rock bottom cylinder pressure would then calculate as:
(Problem_solving_time * Bottom_SAC * 2 * SAC_factor * ambient_pressure_end_of_bottom_time / cylinder_size) + (tank_pressure_consumption_ in_planned_ascent * 2 * SAC_factor)

Example1 (60m, D12, ~25bar for ascent (deco below 21m!), 18barl/min SAC):
(4min * 18barl/min * 2 * 2 * 7 / 24) + (25bar * 2 * 2) = 184bar
Example2 (40m, D12, ~6bar for ascent (direct ascent to 21m), 18barl/min SAC):
(4min * 18barl/min * 2 * 2 * 5 / 24) + (6bar * 2 * 2) = 84bar

The result could be simply printed to the dive plan details together with the info about the corresponding cylinder (for cross check only!).
One could also compare cylinder pressure at end of bottom time with calculated value and give a warning but this IMHO is optional.

I have even a few more key points of the "concept" in my mind like:
- Only do it for OC
- Only do it if depht of last waypoint in dive planner points is deeper e.g. 15m or even 21m (hard coded)
- Concept only will fit simple "regtangular" profiles and very simple multi level dives

atdotde, 2017-02-03:
Hi,

> On 03 Feb 2017, at 21:45, Stefan Fuchs <notifications@github.com> wrote:
> 
> What we would need for the rock bottom calculation:
> 
> Problem_solving_time (including some margin) as GUI input, integer number of minutes, default e.g. 4min
> This is the worst case time the buddy team will stay at max. depth after OOG situation happened.
> SAC_factor
> This is the increase factor for bottom SAC rate in OOG situation. I would multiply bottom SAC by 2 already hard coded and then again multiply by this factor.
> Rock bottom cylinder pressure would then calculate as:
> (Problem_solving_time * Bottom_SAC * 2 * SAC_factor * ambient_pressure_end_of_bottom_time / cylinder_size) + (tank_pressure_consumption_ in_planned_ascent * 2 * SAC_factor)
> 
> 
I think this is a good idea, but let’s do this properly, actually more like in the recreational dive mode. Namely, your formula should be corrected in two respects:

1) Using tank_pressure_consumption_in_planned_ascent assumes you have to do the same deco as in the planned dive. The idea with rockbottom as I understand it is, however, that rockbottom determines the time (better: pressure in your bottom gas) you have to leave the bottom at latest so you could still stop for the problem solving time at the bottom and then ascend.  But if that is earlier or later than in your plan, you might have a different deco obligation. To take this into account, there is no simple formula but you would have to do trial ascends (like in recreational mode). Maybe one could do this iteratively (and hope it converges…), i.e. first use your formula, compute the ascent and its gas use, then apply your formula to the gas use again and recompute the ascent until this stabilises.

2) We work now with real gases. So you shouldn’t simply divide by the cylinder size, but rather use isothermal_pressure (from gas-model.c) instead.

Best
Robert

jbygdell, 2017-02-04:
Hi,
A few things, why the hardcoded duplication of the SAC increase?
Better to just have the SAC_factor and Problem_solving_time as variables that the user can modify.

Do it for all plans, as this also applies to rec dives.

Or simplify it even more, only have SAC_factor as variable and from that calculate minimum tank pressure and by extension maximum problem solving time.

dirkhh, 2017-03-06:
How is this progressing? Since I have started to pull new features maybe this should get merged as well?
Or is it still too early for that?

sfuchs79, 2017-03-06:
Hallo Dirk,

Am 06.03.2017 um 06:41 schrieb Dirk Hohndel:
>
> How is this progressing? Since I have started to pull new features
> maybe this should get merged as well?
> Or is it still too early for that?
>
>
This is the status:
Implementation and review: Code unchanged since a few days. No known
issues. Review once done by Robert.
Testing: Only by me. No feedback yet from Rick, Pablo,...
Documentation: Not yet but I can write s.th. for the user manual during
the next days.
Automated tests: Not yet.

Best regards
Stefan

-- 

Stefan Fuchs
E-Mail: sfuchs@gmx.de <mailto:sfuchs@gmx.de>

dirkhh, 2017-03-06:
Maybe post about it again on the mailing list, asking for feedback. But it may be that I have to pull this before you'll get broader testing.
Yes, a paragraph for the user manual would be good.
And tests. I love things that come with tests.

sfuchs79, 2017-03-07:
On 6 March 2017 at 17:32, Stefan Fuchs <sfuchs@gmx.de> wrote:

> Hallo Dirk,
> Am 06.03.2017 um 06:41 schrieb Dirk Hohndel:
>
> How is this progressing? Since I have started to pull new features maybe
> this should get merged as well?
> Or is it still too early for that? [...]

Sorry for being so slow to do any testing and provide feedback.  In my
limited testing, I have verified that it works and produces expected
results.  On the UI side, I'd consider taking the SAC factor and Problem
solving time preferences out of Gas options, and creating a new group below
called Minimum gas calculation.  But if you'd prefer not to, it's still an
ACK from me.

> Documentation: Not yet but I can write s.th. for the user manual during
> the next days.
>

In the manual entry, it might be worth acknowledging that there are many
ways to calculate minimum gas, and this is just one.  For instance, GUE
advocate adopting a slower ascent rate of 3m/min, compared to 9m/min used
for planning.  This is almost equivalent to tripling the SAC factor that
would otherwise be used; the difference being that it then triples the
problem solving gas use.

> Automated tests: Not yet.
>
>
> Best regards
> Stefan
>
> --
>
> Stefan Fuchs
> E-Mail: sfuchs@gmx.de
>
> _______________________________________________
> subsurface mailing list
> subsurface@subsurface-divelog.org
> http://lists.subsurface-divelog.org/cgi-bin/mailman/listinfo/subsurface
>
>

dirkhh, 2017-03-11:
pulled the patches, so I think we can close this

closed 2017-03-11 by dirkhh