subsurface/subsurface #201: Increase altitude in planner

spoollord, 2017-02-13:
Is there a reason the limit is 3000 on the altitude in the dive planner? I, and my team, regularly dive in the 6000+ altitude. Recommend increasing limit to 10000.

spoollord, 2017-02-13:
P.S. this is in feet.

glance-, 2017-02-13:
The limit is intended for the default(sane) unit meters in desktop-widgets/diveplanner.ui. The limits on the spinbox needs to be updated as well as the unit in desktop-widgets/diveplanner.cpp:197

dirkhh, 2017-02-14:
is a pull request for this coming, @glance- ? We could fix this for 4.6.2

glance-, 2017-02-14:
Patch on the mailing list. git send-email was shorter to write. I should setup hub any year now.

spoollord, 2017-02-14:
Cool. I'll clone and test when I get back from Florida. Lots and lots of wrecks for the next week! :D

PR #204 (2017-02-14): Update altitude limits on unit change
The limits set in diveplanner.ui is intended for the unit set there,
meters. If we move between units we need to update the limits to.

This fixes #201
commit 78059d40 (2017-02-14): Update altitude limits on unit change

The limits set in diveplanner.ui is intended for the unit set there,
meters. If we move between units we need to update the limits to.

This fixes #201
closed 2017-02-14 by spoollord