subsurface/subsurface #224: Debian unable to install - Depends: libssl1.0.0

paddatrapper, 2017-02-24:
I am unable to install on Debian using the Ubuntu PPA because there is an unmet dependency on libssl1.0.0.

apt source:

    deb http://ppa.launchpad.net/subsurface/subsurface/ubuntu zesty main

````
$ sudo apt install subsurface
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 subsurface : Depends: libssl1.0.0 (>= 1.0.0) but it is not installable
E: Unable to correct problems, you have held broken packages.
````

glance-, 2017-02-24:
You're using a ubuntu package on a debian install. Those are not necessarily compatible.

paddatrapper, 2017-02-24:
That's the official way to install on Debian... https://subsurface-divelog.org/download/

glance-, 2017-02-24:
The Trusty binaries works with Jessie. The Zesty packages obviously doesn't work with your debian install which is?

paddatrapper, 2017-02-24:
sid

mturkia, 2017-02-24:
Try using version for older release of Ubuntu. Debian is generally behind in packages when compared to Ubuntu and I could almost guarantee that the libraries on Debian are too old when comparing to a Ubuntu version that is to be released in April. I would try e.g. trusty instead of zesty as is recommended in our Download page.

glance-, 2017-02-24:
Its rather the other way around. Stretch and Sid moved on to openssl 1.1.

paddatrapper, 2017-02-24:
> Its rather the other way around. Stretch and Sid moved on to openssl 1.1.

Yup. Sid is the unstable release of Debian, which has the latest packages as they hit the archive. These are generally ahead of Ubuntu as Ubuntu pulls packages from Debian testing to form their next release.

FTR: using trusty depends on two old libraries: libssl1.0.0 and libzip2

glance-, 2017-02-24:
I'd suggest you go with the Appimage.

mturkia, 2017-02-24:
True, Sid is ahead of Ubuntu, and I don't think there is anything we can do with Launchpad to build compatible packages, any hints are welcome. Well, the binaries are quite likely compatible, but package metadata is based on specific Ubuntu release which apparently does not agree with Sid. So Appimage (or compiling yourself) would be the easiest way ahead - it would be interesting to get a confirmation if the Appimage works on Sid...

paddatrapper, 2017-02-24:
Well the AppImage seems to work. I'll experiment with it and report any issues I have. Only thing I've noticed so far is that it ignores the panels of KDE, so I cannot access the window dressings without first moving the window, but I'll create a new issue for it

closed 2017-02-24 by paddatrapper