subsurface/subsurface #225: AppImage ignoring screen boundaries KDE

paddatrapper, 2017-02-24:
So running the latest AppImage in Debian Sid results in the following:
[image]

It's position is ignoring the boundaries of the screen. I am not sure if this is because I am using 3 screens of different resolutions (1920x1080, 1366x768, 1024x768). Could be a Qt issue

[image]

The top of subsurface is above the top of the screen, which means I must move it down before I can access any of the window dressings or menu bar. Once I can full screen it, full screen works as expected, filling the entire of the visible screen and no more.

dirkhh, 2017-02-25:
That's a pretty hard problem to get right when creating a distro and DE independent image.
The various ways in which KDE and Gnome and enlightenment and Cinamon and ... are incompatible and do odd things when you use Gtk binaries or Qt binaries are one of the reasons why Linux is fundamentally failing as Desktop OS outside of the "extreme enthusiast" market. And given that we bring our own Qt libraries, any hacks done to the OS provided Qt libraries aren't reflected in Subsurface.

It looks from the screenshot like you are on Debian, which is ill-supported by our Ubuntu repos (depending on which Debian version you are on).

I think the best answer here is: You could always just compile from source.

paddatrapper, 2017-02-27:
See #224 for explanation. I filed this bug at the request of @mturkia:
> it would be interesting to get a confirmation if the Appimage works on Sid...

If you don't want to fix it, so be it, though I don't see why you even claim you support Linux, when actually it is only a subset of distributions and any issues that arise you write off as a reason "why Linux is fundamentally failing".

atdotde, 2017-02-27:
The problem with Linux really is that there are a million and one flavor. And they are all different/incompatible in different ways. Plus libraries that are bundled with distributions vary widely (and are often by many months out of date in incompatible ways). The AppImage is really the best effort thing we can handle with the man power available. 

If you really consider this a problem, you should compile yourself (which is not that hard, see https://subsurface-divelog.org/documentation/building/
essentially, you need to install the dependencies, clone the git repository and the rest is handled by a script.

And then we can argue to death if the problem you report is actually Subsurface's or Debian's or the window manager's (hint: it works on other distributions).

dirkhh, 2017-03-06:
Not productive

closed 2017-03-06 by dirkhh