subsurface/subsurface #242: Idea: Change language without to close and restart Subsurface

Ghalr, 2017-03-09:
Hi.

Would be cool to have a feature that prevent to close and restart Subsurface when we want to change language. Some softwares can do it directly.

Woud it be possible ? 
Please. 
Thank you.

atdotde, 2017-03-09:
This seems to be a Qt thing.

dirkhh, 2017-03-09:
On Thu, Mar 09, 2017 at 01:25:36AM -0800, Ghalr wrote:
> Hi.
> 
> Would be cool to have a feature that prevent to close and restart Subsurface when we want to change language. Some softwares can do it directly.

When I implemented the language switcher and read up on how to do this
with Qt it turned out to be extremely hard to unload and reload the
translations - at the time we were using a much earlier version of Qt (I'd
say something like 4.8 or maybe even earlier), so it is possible that this
has been improved.
But the bigger questions is "who would use that?". Outside of the
developers, the typical user doesn't change UI language - at least I
haven't seen anyone make a credible argument about this use case.

Ghalr, 2017-03-09:
Who would use that ?

It can be interesting for developers,translators who are involved in multiple language translation (I grant this is probably not the case here), it is nice to change directly without having to close and reopen the software. Depends of cpu or whatever old computer an user can have it take times... Then probably between foreign divers who do not necessarily speak the other language or who prefers to change rather quickly in his language. Also if a term is not clear in our native language, a fast reminder in english, just go in setting, change, done, will help a lot.

But I agree there is not a credible argument, even if I consider it is only practical and very user friendly. I think it's a great little feature to implement before to have a lot of nice localizations.

neolit123, 2017-03-11:
might be way too much work:
http://www.ics.com/blog/dynamic-language-selection-qt

>It should be mentioned that if you dynamically change the language in your application, a change event of type QEvent::LanguageChange is emitted. You'll have to catch that event and reset the text everywhere in your application.

dirkhh, 2017-03-17:
That's what I recalled. Yes, really cool for developers and translators. But, not really useful for our typical users. And a LOT of work to get right. So I'd go with "no, let's not spend time on this".

closed 2017-03-17 by dirkhh