subsurface/subsurface #248: [Idea] Enhanced annual stats/charts feature.

Ghalr, 2017-03-11:
Hello.

It could be cool to have an enhanced version of "Yearly statistics" using charts such as Diving Log 6.0 tool.
http://www.divinglog.de/blog/?p=132   - outdated pictures.

Thanks.

dirkhh, 2017-03-17:
There was a big push for better stats a while ago and even some code that was written - but the developer got distracted. This, like so many other things, hinges on having someone interested and able to write the code (where "able" means "has the skills and the time").
I certainly would love to see better statistics.

Ghalr, 2017-03-20:
Thank you dirkhh for supporting that idea.
This is clear, both Subsurface and Diving Log 6 are awesome and it'll be great to finally choose the open source version. Better statistics with charts is a thing to have.

Ghalr, 2017-03-23:
General Statistics
Chart Years/Months
Chart Depth range
Chart Temp range (Air/Water)
Chart Dive Time
Chart Dive Start Time
Chart Dive Mode : OC, CCR, pSCR, Freedive
Chart Dive Type/Tags
Chart Dive Location : USA, Europe:France, Europe:UK, Mexico
Chart Visibility : Good (5-4), Average(3), Bad(2,1,0)
Chart Rating : Good (5-4), Average(3), Bad(2,1,0)
Chart Water Type : Salt, Fresh
Chart Entry Type : Boat, Shore
Chart Deco
Chart Repetitive Dive

Dives/ Max. Depth - Dive Time - Air Consumption - SAC - Temp

neolit123, 2017-09-23:
i will leave this issue for @tcanabrava to respond. only labeling for now.
once the new stats functionality is in, perhaps we can take a look at the extended list of requests.

kittykat, 2019-07-01:
I would like to see SAC by exposure suit type and water temperature against time to see how it improves (most of my dives are roughly to the same depth so depth shouldn't be a big deal)

atdotde, 2019-07-01:
Subsurface displays SAC for multiple selected dives in the dive list. If your suit type is in the tags, you can filter on tabs and then compute the SAC for those dives.

kittykat, 2019-07-02:
Do you mean on the statistics tab? That gives me just the average rather than an overview of the trend.

mikeller, 2026-10-17:
Hi @Ghalr,

This has moved a long way since 2017: Subsurface now has a proper chart-based Statistics module (bar, box, pie and scatter series, histograms, a regression-analysis view, and a full legend/axis system), not just the single-table "Yearly statistics" you were comparing unfavourably to Diving Log 6.0 at the time. I haven't individually checked every specific chart type from your wishlist (by water type, entry type, etc.) against the current module, so if something specific from that list is still missing, that'd be worth a fresh, more targeted request.

Closing this as largely addressed by the work since done on Statistics. Please let us know (with specifics) if a particular chart dimension you wanted is still missing.

closed 2026-10-17 by mikeller