subsurface/subsurface #253: Facebook buttons do not work in according to user manual

janmulder, 2017-03-13:
The user manual states "From the _Subsurface_ window it’s easy to determine whether _Subsurface_ has a valid connection to _Facebook_: from the *Main Menu*, select _Share on_ (image *A*, below). Normally, the _Facebook_ option is greyed out. But if there is a connection to _Facebook_, this option is active (i.e. in black color and can be selected)."

This does not work.  Ok, this can be a bug in de user manual, but also a bug in the software.

dirkhh, 2017-03-17:
This worked when I first implemented it. Of course, I never post to FB, so...
yep, you are right. OK, Bug.

Ghalr, 2017-03-25:
I tried Facebook feature Subsurface version 4.6.3, result : WORKING

- I clicked on Share on > Connect to >Facebook  (open the auth form)
- Once connected, give me Disconnect from>Facebook (deconnect work as intended)
- I clicked on Share on>Facebook (open the export form)
- Succesfull "Your dive profile was updated to Facebook."
- Opened my Facebook profile, the dive profile is there.

Maybe to consider few enhanced features:

- Message saying that we are connected to Facebook  ShareOn become Connected to (when connected)
- Message saying that we are disconnected to Facebook  Connected to become ShareOn
- Once the diveprofile is send to the FB profile, new button "open the facebook page" need to insert FB url into Settings or something

- In case that you plan in future to add more social network
ShareOn better to be Social
To connect and open Auth form, we click on the SocialNetwork (ex. Facebook)
Once connected, Facebook is greyout or go into a "Connected to>Facebook"
To diconnect and get a confirmed message, we go into a "Disconnect from >Facebook", Facebook is removed from "Connected to>" and Facebook is not greyout.

janmulder, 2017-03-25:
@Ghalr: indeed as you say "---> Facebook button is not greyout.". This is what this issue is about. I would suggest that you open an new [Idea] for enhancement suggestions (if you like). This way your suggestion will get lost in space when the grayout bug is fixed.

Ghalr, 2017-03-25:
@janmulder Alright I'll do it after dinner time.

PR #317 (2017-04-16): Restore FB grayed out when disconnected
See https://github.com/Subsurface-divelog/subsurface/issues/253. The original behaviour (greyed out when disconnected) is restored. Not sure where and when got lost, but this simple change fixes it.

Fixes #253
commit f2d88619 (2017-04-16): Restore FB grayed out when disconnected

See https://github.com/Subsurface-divelog/subsurface/issues/253. The original
behaviour (greyed out when disconnected) is restored. Not sure where and when
got lost, but this simple change fixes it.

Fixes #253
closed 2017-04-16 by dirkhh