subsurface/subsurface #257: Compile error in marble

g8sqh, 2017-03-17:
Using marble commit 4325da93b, src/lib/marble/TinyWebBrowser.cpp won't compile - error in #include <QPrintDialog>

Toplevel CMakeLists.txt defines QT_NO_PRINTER at line 102, which prevents qabstractprintdialog.h from defining anything.

Commenting out the -DQT_NO_PRINTER at line 102 make marble compile

dirkhh, 2017-03-17:
Hmm.. I build that exact commit on about a dozen different OSs / versions, so I'm a bit surprised by this.

What's the OS / toolchain that you run into this on?

g8sqh, 2017-03-17:
Dirk

I can't see how it could work - so I suspect there's something odd about my system, or QT has changed 

I am an Arch Linux user, so I will have an odd collection of versions

Kernel is 4.8.10
GCC 6.3.1

QT is qt5, v 5.8.0

In a little more detail:

The first two exact gcc errors

In file included from /usr/include/qt/QtPrintSupport/QPrintDialog:1:0,
                 from /home/djch/DIVING/marble-source/src/lib/marble/TinyWebBrowser.cpp:23:
/usr/include/qt/QtPrintSupport/qprintdialog.h:56:1: error: expected class-name before ‘{’ token
 {
 ^
/usr/include/qt/QtPrintSupport/qprintdialog.h:73:20: error: ‘PrintDialogOption’ has not been declared
     void setOption(PrintDialogOption option, bool on = true);

In marble-sources, CMakeLists.txt contains
 add_definitions(-DQT_NO_PRINTER)   at line 102 if we are QT_ONLY (which I am)

src/lib/marble/TinyWebBrowser.cpp line 23 <include QPrintDialog>

which includes qprintdialog.h. At the first error line , the source code is 

class Q_PRINTSUPPORT_EXPORT QPrintDialog : public QAbstractPrintDialog

Earlier qprintdialog.h includes qabstractprintdialog.h, which defines QAbstractPrintDialog at line 55
HOWEVER, this code is protected at line 49 by 

#ifndef QT_NO_PRINTER

Hence, if we have QT_NO_PRINTER defined, the class QAbstractPrintDialog is not defined, and hence all the errors from gcc

Removing add_definitions(-DQT_NO_PRINTER) makes the code compile fine

Happy to supply more info if needed

--David

dirkhh, 2017-03-17:
I'm on Arch as well, so are a couple of the other developers. But it may have been a while since I actually did a clean compile on Arch (that's not one of my build targets, so I don't do a lot of clean builds of the support libraries, just of Subsurface itself).
I'll try this locally and see what's up.
Thanks for the report.

dirkhh, 2017-03-18:
Yep, my guess is this got broken in 5.8 and the other OSs with full builds haven't switched to 5.8, yet.

dirkhh, 2017-03-18:
```
diff --git a/src/lib/marble/TinyWebBrowser.cpp b/src/lib/marble/TinyWebBrowser.cpp
index 52e88422c7bd..0d86f8b0853f 100644
--- a/src/lib/marble/TinyWebBrowser.cpp
+++ b/src/lib/marble/TinyWebBrowser.cpp
@@ -20,8 +20,10 @@
 #include <QAction>
 #include <QDesktopServices>
 #include <QPainter>
+#ifndef QT_NO_PRINTER
 #include <QPrintDialog>
 #include <QPrinter>
+#endif
 #include <QTextFrame>
 
 // Marble

```

That fixes it for me.

dirkhh, 2017-04-14:
This should be fixed now.

closed 2017-04-14 by dirkhh