subsurface/subsurface #258: Subsurface-4.6.3-x86_64.AppImage SIGSEGV during startup

g8sqh, 2017-03-17:
Current appimage crashes after the window is created. Under gdb I get 
... 
Thread 16 "Thread (pooled)" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffcd7fa700 (LWP 2125)]
0x00007ffff7de596c in _dl_relocate_object () from /lib64/ld-linux-x86-64.so.2
(gdb) bt
#0  0x00007ffff7de596c in _dl_relocate_object () at /lib64/ld-linux-x86-64.so.2
#1  0x00007ffff7ded891 in dl_open_worker () at /lib64/ld-linux-x86-64.so.2
#2  0x00007ffff012ec74 in _dl_catch_error () at /usr/lib/libc.so.6
#3  0x00007ffff7ded0a9 in _dl_open () at /lib64/ld-linux-x86-64.so.2
#4  0x00007fffeddedf86 in  () at /usr/lib/libdl.so.2
#5  0x00007ffff012ec74 in _dl_catch_error () at /usr/lib/libc.so.6
#6  0x00007fffeddee669 in  () at /usr/lib/libdl.so.2
#7  0x00007fffeddee022 in dlopen () at /usr/lib/libdl.so.2
#8  0x00007ffff0ea1b50 in  () at /tmp/.mount_Zl72CU/usr/lib/libQt5Core.so.5
#9  0x00007ffff0e9cf04 in  () at /tmp/.mount_Zl72CU/usr/lib/libQt5Core.so.5
#10 0x00007ffff1e0123a in  () at /tmp/.mount_Zl72CU/usr/lib/libQt5Network.so.5
#11 0x00007ffff1e01829 in  () at /tmp/.mount_Zl72CU/usr/lib/libQt5Network.so.5
#12 0x00007ffff1df44ce in  () at /tmp/.mount_Zl72CU/usr/lib/libQt5Network.so.5
#13 0x00007ffff0d01a2d in  () at /tmp/.mount_Zl72CU/usr/lib/libQt5Core.so.5
#14 0x00007ffff0d049a7 in  () at /tmp/.mount_Zl72CU/usr/lib/libQt5Core.so.5
#15 0x00007ffff6b062e7 in start_thread () at /usr/lib/libpthread.so.0
#16 0x00007ffff00f554f in clone () at /usr/lib/libc.so.6

Kernel is 4.8.10 .  md5sum is 
d9b7e9fe7a6b880f890ba444da6dfbbe  Subsurface-4.6.3-x86_64.AppImage

whole gdb session attached
[sub.txt](https://github.com/Subsurface-divelog/subsurface/files/850314/sub.txt)

Happy to supply further info if needed. (I have built from source - so not urgent for me, maybe for others)

dirkhh, 2017-03-17:
What Linux distro is this? We have had success reports from quite a few different ones at this point, so I'd love to be able to reproduce the problem

neolit123, 2017-09-23:
closing due to lack of user feedback.

closed 2017-09-23 by neolit123