subsurface/subsurface #264: Incorrect dive time - Aladin Tek 2G

iainmsmith, 2017-03-19:
Dive to 27m for 38min on an Aladin Tec2G downloads into Subsurface as 9min 35s.
Downloading the dive into Uwatec's SmartTRAK software shows the correct dive time.
Downloading the Tec2G into a fresh Subsurface logbook produces a list of dives for which _all_ of the times are wrong.

This appears to be a new problem, as I have not noticed this before (last "good" download was in Nov 2016). An imported Liquivision .lvd data file for the same dive contains the correct data. 

The .xml file containing two erroneous Tec2G downloads and the corresponding Xen logbook import is attached as a .txt file.
Tec2G dive 1 corresponds to Xen dive 46, Tec2G dive 2 corresponds to Xen dive 47.

OS: Win10
Subsurface: 4.6.3
(the problem was originally noticed with 4.6.0, but upgrading to 4.6.3, deleting the dive and re-importing reproduces the same error).

[Tec2G Error.txt](https://github.com/Subsurface-divelog/subsurface/files/853489/Tec2G.Error.txt)

atdotde, 2017-03-19:
Hi Iain,

> On 19 Mar 2017, at 21:13, iainmsmith <notifications@github.com> wrote:
> 
> Dive to 27m for 38min on an Aladin Tec2G downloads into Subsurface as 9min 35s.
> Downloading the dive into Uwatec's SmartTRAK software shows the correct dive time.
> Downloading the Tec2G into a fresh Subsurface logbook produces a list of dives for which all of the times are wrong.
> 
> 

it seems, we parse the sample rate wrong by a factor of 4. That shouldn’t be too hard to find. Could you please mail a dump file as well (select the corresponding checkbox in the download dialogue)?

Thanks.

Best
Robert

iainmsmith, 2017-03-19:
Hi Robert,

Dumpfile en route to you by email to your atdotde address.

Iain

dirkhh, 2017-04-22:
Has this been addressed?

atdotde, 2017-04-22:
Jef addressed it in his libdivecomputer commit 640308c0763ae263f2e4231b805794de585a6fc6. We need to cherry pick this (as well as maybe some others) to the Subsurface-branch.

janmulder, 2017-04-22:
This commit is already merged in the Subsurface-branch. No cherry pick required.

neolit123, 2017-09-23:
closing as this was fixed in libdc.

closed 2017-09-23 by neolit123