subsurface/subsurface #271: [Idea] Add Carbondive cylinders / Pony bottles

Ghalr, 2017-03-23:
Hello.

Would it be possible to add Carbondive cylinders (6,8L and 12L @300bar -10L not produced anymore(.?)) and also consider to add Dual Set for them ?

Would it be possible to add Ponny Bottles ?
example : Catalina Ponny Bottles 
http://www.leisurepro.com/p-aqutp/catalina-pony-bottle-tanks-yellow-with-pro-valve
Available 6, 13, 19, 30, 40 cu.

I have located the file at subsurface/core/equipment.c

glance-, 2017-03-23:
I think we should publicize that the drop down gets populated with cylinders from you logbook to, so just add it in one dive, by hand, and you can choose it from the drop down after that.

dirkhh, 2017-03-24:
I agree. This is tricky. Adding more and more cylinders doesn't make things easier for the user...
If the dropdown has 50+ entries, it's pretty much useless.

Ghalr, 2017-03-24:
Right now I can consider 2 things:
- Cyl. we bought = presaved gear (default)
- Cyl. we rented = manually add and presave if we dive a lot with that cyl. size
use a presaved list (cyl. we bought + rented) use 2 colomns to choose which cyl. size we want to be into that presaved list, then remove when the trip is done. "Move" and "Remove" options
If Subsurface is used by divers who keep every detailed divelog, those data are important for stats and a serious planification /debriefing

Bocio, 2017-04-09:
I copy this from the list:

I still see this phantom AL7 thank with 10 liter capacity. The item on the list is AL7 but the chosen value is AL72. Actually browsing the list there is another AL72 and the chosen description correspond with the dropdown item list. I think this is a very old bug. My two years old screencast is still valid. (Browser complaint about a invalid certificate but it's safe)

https://www.screenr.com/kTEN

I found this while trying to change my default cylinder. By default I find a D7@300bar but I have a D7@200bar so I manually add this to one of my dives. Unfortunately I do not find it in the preference list of default cylinder.

Do you populate that list with only hard-coded cylinders?

@Ghalr idea is nice. I'm the only one I find counter-intuitive inserting a new tank from a dropbox?

dirkhh, 2017-04-22:
The user experience with the cylinders is still rather frustrating to me. But as much as I tinker with the code, I can't make it do what I want it to do. The search doesn't always start when I type. Sometimes changes don't stick. It's just not a great experience.

I really wish that a more experienced Qt developer could try to make the equipment UI better.

Ghalr, 2017-04-23:
If I can give my point of view and as I said earlier it would be great to categorize those cylinders.
- common US/EU cylinders types
- common used cylinders size, 

and it will give user to choose where each cylinders belongs to.

- purchased cylinders
- rented cylinders
- archived cylinders

Add a new cylinder option who can go by default to "Archived"

These cylinders can move freely with a right - left button where user can move cylinders according to the needs. So if diver most often uses the purchased cylinders, he will pickup the selected cylinders into "Purchased" box, and during a trip he can choose a cylinder into "Rented", and once the trip is done, unused cylinders / or rented cylinders will go into "Archived" box.

Bocio, 2017-04-24:
> That said, the underlying request (add more default cylinders) I don't agree with. We should have a VERY small set of default cylinders, and otherwise users should add their own cylinders which will be offered next time you add a dive.

I see your point but currently it doesn't work. It was the first route I tried. Mika gave the same suggestion on the list but it doesn't work. So I asked to add this cylinder in the default list as last resort. 
BTW in default list there is a D7@300 bar. I never saw that kind of tank in 30 years of diving.

Tank management is so a mess that I had to switch to CCR  :-) but even now I have to insert manually my default tank for every dive. OMG.

Bye

Bocio, 2017-04-24:
Dirk, same old story with you:
http://www.wordreference.com/ende/touchy

The more you reply touchy the more I like to tickle you :-)

closed 2017-04-24 by dirkhh