subsurface/subsurface #273: subsurface-mobile cloud authentication does not work

berkovskyy, 2017-03-24:
After I install subsurface-mobile application I try entering my credentials (which I used in desktop app) but I get an error message "No response from cloud server to validate the credentials" :
[image]
I tried this couple of time during the day and I checked that my internet connection is working before every attempt.

Interestingly I can enter my other email and receive authentication PIN via email, but the app reports exactly same error and there is no field where I can enter the PIN.

I am running subsurface-mobile version 1.2 on latest iOS (10.2.1) on iPhone 6.

dirkhh, 2017-03-29:
The account that I can guess from the screen shot hasn't been verified. I don't quite understand why it isn't offering you the option to enter the pin. The screen shot shows that you are on wifi. Do you happen to be behind a proxy?
One thing to try would be to use Subsurface on a laptop, enter the email/password above. That should prompt you to enter then PIN which enables your account - but that still doesn't explain why you don't get a response from the cloud server.

Ghalr, 2017-03-31:
I tried to register on subsurface-mobile yesterday for the first time, got the PIN, and it worked.
I used a @gmail adress.
Not behind a VPN/Proxy; or tried to use it in PC version first.

dirkhh, 2017-04-04:
@berkovskyy - any update on this?

dirkhh, 2017-04-14:
Closing with no update

closed 2017-04-14 by dirkhh