subsurface/subsurface #274: Wrong pO / pN strings (in english) and others ?

Ghalr, 2017-03-24:
Hello.

Scrolling down Transifex to hunt some errors, I have found those:

pO%s: %.2fbar⏎    Occurrences: ../core/profile.c:1319
pN%s: %.2fbar⏎    Occurrences: ../core/profile.c:1321
Apnoea                 Occurrences: ../desktop-widgets/configuredivecomputerdialog.ui:2346
Dive log files [...] 87words  Occurrences: ../desktop-widgets/mainwindow.cpp:1832

Should be pO₂, pN₂, Apnea, and for the last stings, is it correct to have a lot of words ?

janmulder, 2017-03-24:
pO%s: %.2fbar⏎ ../core/profile.c:1319
pN%s: %.2fbar⏎ ../core/profile.c:1321 : Are correct: the first %s is filled with UTF8_SUBSCRIPT_2 at runtime, so a nice subscripted 2.

Apnoea ../desktop-widgets/configuredivecomputerdialog.ui:2346 : Correct (alternative) spelling according to https://en.wikipedia.org/wiki/Apnea.

The & in menu items denote the location of the shortcut character (so no error).

And, finally, unsure what you think is wrong with Dive log files [...] ../desktop-widgets/mainwindow.cpp:1832, Ok, looking from transiflex it seems a strange piece of text, but it is a valid string to fill an option menu from code.

Ghalr, 2017-03-24:
All my apologizes about the pN / pO strings, I still think for consistency to keep Apnea and for the other.
To translate Personali&ze (in French) it will be Per&sonnaliser So I have no idea where exactly & is supposed to be.

atdotde, 2017-03-24:
The & marks the character for a keyboard shortcut (with alt). S might not be such a wise choice.

Ghalr, 2017-03-24:
Thank you to confirm that.
The translation was made by Guillaume_G, a year ago
[If you are around Guillaume_G ?]
in any case we do not have Z in our french translation:
- Personnaliser
- Personnalisation
- Custom

Maybe I should create a list of "keyboard shortcut" to translate better ?
Do those "keyboard shortcut" are activated ?

janmulder, 2017-03-24:
To see this type of keyboard shortcut, press and hold the alt key. The chosen letter gets an underline.

Ghalr, 2017-03-24:
This is perfect. Thanks.
Well if everything is finally fine (sorry again for the bug ticket) feel free to close it, I'll have a closer look into the translation later.

Ghalr, 2017-03-25:
@janmulder 
About what you said for pO% and pN% (filled with UTF8_SUBSCRIPT_2 at runtime, so a nice subscripted 2) why not use this method for the others too ?

janmulder, 2017-03-25:
why not use this method for the others too? 

Well, why would we change the code for at least 30 strings, when there is no problem? There are more ways to display the subscripted 2's, and the UTF8_SUBSCRIPT_2 is just a way. We could argue about consistency, but in that case I would change the 3 occurrences of UTF8_SUBSCRIPT_2, as it is no use to dynamically construct that part of the string.

And just added a pull request for the Apnoea change.

Ghalr, 2017-03-25:
You marked a point about to just change the 3 occurrences of UTF8_SUBSCRIPT_2

janmulder, 2017-03-25:
@Ghalr: not sure what you mean?

Ghalr, 2017-03-25:
You are right to change only 3 strings than 30 strings, I'm agree with you.

janmulder, 2017-03-28:
@dirkhh: can you close this issue?

closed 2017-03-28 by dirkhh