subsurface/subsurface #279: [Idea] Enhanced features for social network sharing (and divelog online)

Ghalr, 2017-03-25:
Hello.

As suggested by @janmulder I would like to request some enhanced features for social network sharing.
Social network I'm talking about: Facebook, WhatsApp, Instagram, Tumblr
Online Dive Log I'm talking about: [https://en.divelogs.de/](divelogs.de), [https://logbook.diviac.com/](Diviac), [https://diveboard.com/](DiveBoard)

How to connect ?
- Preferences>SocialNetwork  such as Subsurface cloud storage form
- Share on>Choose SocialNetwork / Choose OnlineDiveLog
- Once connected , buttons are greyed out.
- Disconnect from>All or Specific SocialNetwork / OnlineDiveLog

Enhanced feature:
- Date/Time/Duration/Notes should be share as default
- If no divelog is selected, Facebook should not work

Ghalr, 2017-03-25:
#280

Ghalr, 2017-03-25:
#253

dirkhh, 2017-03-29:
Thanks for the suggestions.
Here are my thoughts:
- treating online dive logs and social networks the same seems wrong. With an online divelog we'd want a bidirectional connection (as we have for divelogs.de). A social network is "post only".
- while it shouldn't be all that hard to create Instagram/Tumblr/Twitter integrations, we are back to my standard answer... is there a developer interested in doing this?
- same is true for the other online dive logs, of course (but that tends to be more work because one would want this to be bidirectional)

closed 2017-03-29 by dirkhh