subsurface/subsurface #285: planner: save new copy of plan

janmulder, 2017-03-26:
Plan a dive with the planner. Re-open the dive in the planner, do a little edit (I just changed the depth with 1m), and save the plan as a new dive/duplicate. Now 1 occurrence in the divelist has a filled-in test block (with the planned deco/gas use data from the planner), but the other occurrence has an empty text block. 

Seems a bug to me.

sfuchs79, 2017-03-26:
I can confirm that I also saw this behaviour once a few weeks ago when playing a lot with the planner. But then when I tried to reproduce it again it did not happen any more. It simply disappeared.
@janmulder Do you have a setup/file where you can reproduce this again and again? As a side note: I for sure wont't be the right person to fix this but I'm curious because as I'm saying before I also discovered this.

janmulder, 2017-03-26:
@sfuchs79: I just tried it on an empty, new logbook, and it reproduces there immediately. I found it initially in my regular logbook, but that one is big. 

Running latest master 4.6.3 (155)

sfuchs79, 2017-03-26:
Hihi, that is funny...
When trying to reproduce this very quickly I got this result: Replanned a dive two times and for the replanned dives I have double results in the notes (see file). But no empty notes.
[test_replan.zip](https://github.com/Subsurface-divelog/subsurface/files/870935/test_replan.zip)

Yes, there is a bug.... ;-)

atdotde, 2017-03-27:
Here is what the code says:

			QString oldnotes(current_dive->notes);
			if (oldnotes.indexOf(QString(disclaimer)) >= 0)
				oldnotes.truncate(oldnotes.indexOf(QString(disclaimer)));
			if (did_deco)
				oldnotes.append(displayed_dive.notes);
			displayed_dive.notes = strdup(oldnotes.toUtf8().data());

In prose: Does the old notes contain the word "disclaimer" (i.e. are those notes produced by the planner)? If yes, get rid of everything after and including the word "disclaimer". 

Then, does the new plan have deco stops? Then add the new notes to what remains of the old notes.

Is this what you are experiencing? Then its a feature, not a bug.

atdotde, 2017-03-27:
Ah no. It's not the word "disclaimer" but the actual (possibly translated) disclaimer. And since that now contains the name of the deco model (untranslated, bummer!), it does not recognize it if the model is changed.

janmulder, 2017-03-28:
Ok, back to the question "Is this what you are experiencing? Then its a feature, not a bug.". 

I am (now) testing on the atdotde:notesinreedit branch (I pulled it to my local environment)

I see two situations: 1) replan and still no deco, 2) replan and now deco

ad 1) This is what I did when I wrote my issue. Lets be more exact in what I am seeing.  Plan a dive 20min at 15m (the default dive). Now replan this and change it to 20min at 16m, and save as duplicate. I am now in the terrain that @atdotde calls a feature, not a bug. However, the original 20@15 dive has the notes filled in, and the replanned 20@16 has empty notes. If is was the other way around (the replanned dive has the data filled in, the original not any more), I might understand "the feature", but for this I can see no logic. 

ad 2) now deco on the replan. This works as I expect. Filled in on both occurrences. In fact, this results in the same remark @sfuchs79 make: why the "in deco" difference? I think I do not see the rationale for that either. 

side note: what can be a little confusing, is that the dive list (in default ordering, displays the original (first planned) on top and the replanned as second entry.

PR #289 (2017-03-28): Handle notes in replan
Upon replanning a dive, we want to delete the old
dive plan in the notes and replace it with the actual.

This fixes a problem when we failed to detect the old plan due
to the deco model name appearing in the disclaimer that was used
as a marker for the notes.

This patch also adds translation markers for the deco model name strings..

Fixes #285
commit eefa390e (2017-03-28): Handle notes in replan

Upon replanning a dive, we want to delete the old
dive plan in the notes and replace it with the actual.

This fixes a problem when we failed to detect the old plan due
to the deco model name appearing in the disclaimer that was used
as a marker for the notes.

This patch also adds translation markers for the deco model name strings..

Fixes #285
closed 2017-03-28 by dirkhh