subsurface/subsurface #286: tank data not displayed when importing from Uwatec/Memomouse

olf, 2017-03-26:
I'm reading data from an old Uwatec Aladin Air through the Memouse. It works quite well, except for one detail: the tank data is not shown in Subsurface. I've checked the XML that I received through dctool and the tank data is indeed in a somewhat strange format:

```
<tank>
   <gasmix>0</gasmix>
   <beginpressure>169.61</beginpressure>
   <endpressure>0.00</endpressure>
</tank>
```

i.e. this is basically the consumption, not the start and end pressure. endpressure is always 0.00, for all dives in the log file.

Here's a full dive record, without profile data:

```
<device>
<dive>
<number>1</number>
<size>18</size>
<fingerprint>DEADBEEF</fingerprint>
<datetime>2013-03-19 13:44:55</datetime>
<divetime>60:00</divetime>
<maxdepth>17.81</maxdepth>
<temperature type="minimum">21.8</temperature>
<gasmix>
   <he>0.0</he>
   <o2>21.0</o2>
   <n2>79.0</n2>
</gasmix>
<tank>
   <gasmix>0</gasmix>
   <beginpressure>139.27</beginpressure>
   <endpressure>0.00</endpressure>
</tank>
</dive>
</device>
```

I think it should be be OK to display the values just like that (i.e. as start and end pressure) in Subsurface, as they would still enable AMV calculations etc. (after adjusting the tank size etc.) Currently, the consumption data is lost.

There's another oddity with these records, unrelated to the tank data: the time of off by ~85 minutes, i.e. the dive above was at 15:09. But that's not Subsurface's fault, of course.

dirkhh, 2017-03-29:
our code considers the end pressure of 0 as invalid. I should be able to add an exception that allows this as valid for the memomouse.
We used to be able to import the XML that the libdivecomputer command line tool produced, but clearly that code hasn't kept up with the latest changes in dctool. I'd consider that a bug (but that's only tangentially related to this issue).

dirkhh, 2017-03-29:
I'm building test binaries that should fix this when downloading from the memomouse. What OS are you running (so I can point you to the correct daily build)?

olf, 2017-03-29:
I'm on MacOS 10.12.4 currently. Happy to test. :-)

dirkhh, 2017-03-29:
> I'm on MacOS 10.12.4 currently. Happy to test. :-)
> 

Can you please try http://subsurface-divelog.org/downloads/daily/Subsurface-4.6.3-169-g09b5f9b99c96.dmg <http://subsurface-divelog.org/downloads/daily/Subsurface-4.6.3-169-g09b5f9b99c96.dmg>

olf, 2017-03-29:
I tried that version and unfortunately, it doesn't work as expected.

I've done these steps:

- File > New logbook
- Import > Import logfiles
- used this file: [memomouse_excerpt.xml.zip](https://github.com/Subsurface-divelog/subsurface/files/879703/memomouse_excerpt.xml.zip) - it contains 2 dives: one with profile, one without.
- check the "Equipment" tab on both dives: no tank data is available

dirkhh, 2017-03-29:
I think I misunderstood what you were trying to do. What I fixed (I hope) was the direct download from the Memomouse using Subsurface - so not the XML import. That's on my todo list, but it won't happen today...

olf, 2017-03-29:
Ah, ok. :-)

I just tried the download and that works indeed. However, due to end pressure being 0.0, if I specify the tank size, it won't calculate SAC etc. I need to set end pressure to e.g. 1.0, then things are working as expected. I guess that's OK for a workaround. Or would it be possible to allow 0.0 as end pressure? Or maybe 0.1 (minimum value I can use is 0.6, Subsurface will discard values of 0.5bar or lower).

dirkhh, 2017-03-29:
The problem is that 0 bar fundamentally is an invalid pressure. And we therefore recognize it in several spots as "special".
What would be much easier, is to shift that arbitrary range that the Memomouse gives you. Instead of N..0, could we do (N+30)..30 ? This way you'd get a much more "normal" looking graph with likely much more realistic numbers. Would this be a workable solution for you? It's a bit of a hack - but less of a hack then what the Memomouse gives us in the first place.

olf, 2017-03-29:
Sure, shifting by e.g. 30bar would be fine. It's all about the consumption anyway, not about the absolute start/end values. So, yeah, it'd be awesome, if you could implement it that way.

dirkhh, 2017-03-30:
Next try: http://subsurface-divelog.org/downloads/daily/Subsurface-4.6.3-170-g34a42d7f0431.dmg
This shifts the tank data up by 30bar. You'll need to download from the Memomouse again.

olf, 2017-03-30:
Works like a charm. Thanks a lot. 👍 😃

dirkhh, 2017-03-30:
Thanks for reporting backl! This will be in our next release, but we haven't scheduled that, yet. So please use the daily build until then.

commit 09b5f9b9 (2017-03-29): Accept 0 as valid endpressure for Uwatec devices

Instead of delivering the actual start and end pressure, memomouse
gives you a start pressure that matches the delta between actual
start and end pressure, and an end pressure of zero. Who the heck
knows why it does that, but the information is better than nothing,
so we should accept it.

Fixes #286
closed 2017-03-30 by dirkhh