subsurface/subsurface #294: [Idea] Option (checkbox) to download beta version

Ghalr, 2017-03-31:
Hello.

I know this website http://subsurface-divelog.org/downloads/daily/?C=M;O=D
to download daily updates, and I wonder if it would be possible to have a checkbox to download those beta daily version. I understand that developers are already up-to-date while coding, but it would be great for users to do it also. I mean to download beta version.

Also I wonder if it's not better to wait stable release, ?

Regards,

dirkhh, 2017-04-22:
I thought about this... and we do actually offer to download new beta versions, if you have a beta version installed, but we never offer to download daily builds - and I think that's a good thing. It takes some rather serious commitment to actually test these not-daily builds... and things sometimes get broken pretty badly in them. Maybe a link to downloads/daily from the About box? That might be doable. But even that is likely to get people to use those dev builds that don't quite understand what they are signing up for...
I'm leaving this open for more discussion, but as you can tell by the label, I'm not convinced.

Ghalr, 2017-04-22:
I do understand. Since 3 weeks I have made around 12 dives who are still not imported into Subsurface.
Weekly builds would be more interesting than daily builds for users who are using more often Subsurface or who are developers.

Anyway it would be great to have a clickable lilnk into "About..." windows to let users know that it's possible to download a more updated Subsurface. Latest stable release was on 2 march and there are a lot of bugfixes and new features who are really interesting.

dirkhh, 2017-04-22:
Hi @Ghalr  - I am planning to make a 4.6.4 fairly soon.
Those 12 dives - are you saying they can be imported with the latest daily build but not with 4.6.3, or can they not be imported by either?

Ghalr, 2017-04-22:
Sorry I had no time to use my laptop to import them but I believe everything is good.

dirkhh, 2017-04-22:
It would be great if you could check this when you have time - because if there's something that's broken, we should fix it before releasing 4.6.4. Thanks

Ghalr, 2017-04-22:
Will do this Monday.

Ghalr, 2017-04-22:
Everything are fine Sir

dirkhh, 2017-04-23:
Thanks for checking - I really appreciate it.

neolit123, 2017-09-23:
closing this one as we won't add the "beta download" option.
@Ghalr please re-open another issue if needed.

closed 2017-09-23 by neolit123