subsurface/subsurface #297: Problem with extra "ghost" cylinders on logs generated from Cobalt 2 dive computer

idaholion, 2017-04-04:
I just OW certified, and am checking out various dive log programs. I own a Cobalt 2 computer, and used the Cobalt Logbook program to download the dive logs off of the Cobalt. I then opened up the logs in Subsurface 4.6.3, and am trying to figure out how everything works. My first dive day was on Air, and everything looks good. My second dive day was on Nitrox. The filkes look fine on the Cobalt Logbook program, but on the Subsurface, I have a phantom tank keeping the air pressure from being graphed. The screen looks like this
[image]
The pinked out square showing 2091 psi is on each of the second day dives. I tried deleting it, it won't go away. And I notice that there is no tank pressure being graphed. 
I tried downloading the files from my wife's loaner cobalt, and all of her files have this same pink pseudo tank (different PSI on hers, she has 3 different pink PSI numbers).  The lines with the "pink" start pressure and end pressure squares cannot be deleted UNLESS I go in and manually change the information in the O2% line. As far as I can tell, this value HAS to be the same as the other line on the dive. I can change either line, the line with the pink squares or the line without. As long as both lines have the same O2% value, then the "remove this cylinder" square at the beginning of each line is enabled, I can delete the line, and the air pressure line appears on the graph. I also discovered why the squares were pink. If there is not a number in the Start Pressure AND the End Pressure data locations with the End Pressure less than the Start pressure, the 2 squares turn pink.

dirkhh, 2017-04-04:
I'm a little confused by your description. Did you download from the Cobalt using Subsurface, or did you download using other software and then imported the data into Subsurface?
Most likely you have multiple tanks defined on the Cobalt with the same gas mix and Subsurface doesn't know which one of them you were actually breathing from. But we should be able to simply take the one with the correct (I assume) start and end pressure. But I need to understand a bit better what's actually happening here...

idaholion, 2017-04-04:
The one thing I haven't tried yet is downloading directly from the Cobalt computer to the Subsurface program. This problem has appeared when importing a log file downloaded from the Cobalt computer to the Atomic Cobalt logbook program, then "import log files" from the cobalt logbook saved file to the Subsurface program. 

The problems came off of dive logs downloaded from 2 different Cobalt dive computers. It only was an issue on the Nitrox dives from my Cobalt, but all of the dives, both Nitrox and Air, were affected on the logs from my wife's computer. I will have to check on the settings of the dive computers, but it is a bit odd for 2 different dive computers to have the same problem of settings.  On my dive computer, the problem did not exist when I had the setting set to Air, only when I shifted over to Nitrox.  I have reviewed my Cobalt, I can't find a location where I could have multiple tanks simultaneously defined.  I also can not figure out where the value in the "extra" cylinder (the 2091 PSI in the example) came from on that dive.  
I used those exact dive logs from the Atomic Logbook program and transferred them to Diving log 6.0.  As far as I can tell, in Diving Log 6.0, everything works fine, no "phantom" cylinder information.
I am open to suggestion.

idaholion, 2017-04-05:
I came home and checked the setting on my Cobalt.  There is a setting called Gas/Tank settings.  In that setting there is one tank listed as Primary (on hose)  The current primary tank is a EAN33 80cf@300PSI.  There are 2 more that are listed as additional mixes which can be shifted to primary as desired.  No way that I can see to have multiple tanks defined as active.
Additional information.  The 2091 PSI pressure in the Phantom Cylinder is the starting pressure of an earlier dive, the third dive on the computer.  It is the "phantom" pressure in the 5 subsequent dives, dives 4,5,6,7 and 8.  The first 3 dive logs all have a phantom pressure of 3029 psi on the dive log. Since this was a borrowed Cobalt, I assume that the 3029 PSI value came from an earlier log that I did not download.

The logs I downloaded from my brand new cobalt had no phantom logs for the first 4 logs.  The starting pressure for the 4th log was 1325 PSI.  The next 4 logs all have phantom cylinders with 1325 PSI as a start value.
Each of the Phantom cylinders have carried over values for Size, Start Pressure, Bot MOD, and MND values. By carry over values, I mean that it carried over these values from dive 4 to dives 5,6,7 and 8.   All the rest of the values are blank.  The only way to delete a "phantom cylinder" value is to manually input the value from the other line for O2%.  It has to be the same value as the O2% value from the line below it.  As soon as that value is input, the "clicking here will remove this cylinder" icon is active.  
S in this example, this is dive 4
[image]
Then this is dive 5
[image]
Unless I manually put in 33% in O2%, I can't delete the extra line and the pressure curve will not appear on the chart..
I hope this information is helpful to determine what is happening.

mturkia, 2017-04-10:
Can you send me sample logs of what you imported to Subsurface? There is probably something I have not seen before in them, thus not handled properly.

idaholion, 2017-04-10:
Sure, just send me your email address

idaholion, 2017-04-10:
Attached is one of the log files.

This log file has 8 dives, all of them with phantom cylinders in the equipment tab in Subsurface.
The start pressure on the 3rd log is the "phantom" cylinder pressure on logs 4-8

mturkia, 2017-04-11:
For some reason, I fail to see the attachment. Maybe the email integration of github is ignoring that part. Anyway, you can send it directly to miika.turkia (at) gmail.com

closed 2017-04-12 by dirkhh

idaholion, 2017-04-12:
Sorry, I'm afraid I don't know what this means.  I recognize that there is a change identified, but I don't know if this is a change that will be enacted in some future revision, or if this is just listed as a bug for future review, or exactly what change #302 actually indicates.  I am afraid I am a GitHub novice.

Closed #297<https://github.com/Subsurface-divelog/subsurface/issues/297> via #302<https://github.com/Subsurface-divelog/subsurface/pull/302>.

dirkhh, 2017-04-13:
What this means is that we have checked a commit into our master branch that we believe will address the issue that you are reporting. So this should be addressed in an upcoming "daily" build (misnomer - they happen roughly whenever I feel like it... I'm planning to do one later today) and then in our next release.

idaholion, 2017-04-17:
I tried the latest fork version, It does seem to solve my problem.  Thank you

Somehow in this process I have apparently selected to follow new issues, and am now getting emails on several topics on a daily basis.  This was done inadvertantly.  How do I get off the email notification list for these other issues?

Thank you

PR #302 (2017-04-12): Ignore empty cylinders on Cobalt import
It appears that Cobalt might include additional gas mixes, and only way
to determine what is used appears to be to ensure that start and end
pressures are greater than 0. One would assume there to be something
else available in the database, but I was not able to spot it.

Fixes #297
commit e355037e (2017-04-12): Ignore empty cylinders on Cobalt import

It appears that Cobalt might include additional gas mixes, and only way
to determine what is used appears to be to ensure that start and end
pressures are greater than 0. One would assume there to be something
else available in the database, but I was not able to spot it.

Fixes #297