subsurface/subsurface #301: Petrel 2 Deco model info

janmulder, 2017-04-10:
Imported a dive from a Petrel 2, firmware version 38, set on VPM-B/GFS +2 with GFS 85%. Using version 4.6.3. 

The extra information tab is containing garbled data (from the xml):
<extradata key='Deco model info' value='VPM-B/GFS +2 %85??6?\330\352' />

This disables the reload of the xml/ssrf  (but for some reason allows storage and retrieval from the cloud), with a message like ".../test.xml:83324: parser error : Input is not proper UTF-8, indicate encoding !
Bytes: 0xC8 0x37 0x30 0x3F
  <extradata key='Deco model info' value='VPM-B/GFS +2 %85�70? ��??' />"

I do not have direct access tot the Petrel (I am posting this for a buddy).

glance-, 2017-04-10:
The git storage backend (which is used by the cloud storage) is way more resilient against crap data than xml is.

Anyhow, i can't really figure out whats happened here.

The line that generates that data is:
https://github.com/Subsurface-divelog/libdc/blob/Subsurface-branch/src/shearwater_predator_parser.c#L412

(note that the % is moved in later releases to the propper place)

PeterZaal, 2017-04-10:
Hi, I am the owner of the Petrel (with the VPM software). Is there anything I can help you with to determine the cause? The problem is now that is not possible to create a working local xml/ssrf copy of the cloud storage data.
E.g. do you need any files from the local cache of the cloud storage? Anything else?

glance-, 2017-04-10:
You could create a memory dump of your device, and email that one to me, and i'll try to reproduce the problem.

On which platform are you running Subsurface? Something with a weird snprintf?

glance-, 2017-04-10:
And, can you reproduce the issue when downloading into a clean logbook? Or was it a one-time issue?

glance-, 2017-04-10:
And, you can always workaround the issue by hand-fixing the xml. Just remove the crap, save the file, open it in subsurface and save to the cloud, and the issue should be gone for now.

If it was a one time issue that might be the end if it, but if you can reproduce the issue it wasn't due to random memory corruption, but rather due to a bug somewhere.

PeterZaal, 2017-04-10:
Hi, thanks, I will try your suggestions tonight, and let you know. Btw I am using Windows 10 Home.
The weirdest thing is that the % (of the GFS) is still in front of the number, although I am certain that I am running 4.6.3 where the % should be after the number. This is the part of the text in the divecomputer file: "Deco model info" "VPM-B/GFS +2 %85È70 ‡€"
So where are the extra characters coming from..?
Will let you know more tonight.

dirkhh, 2017-04-10:
There are two questions in my mind:
a) does this happen with a fresh import, using Subsurface 4.6.3 (our latest version)
b) would you like me to just fix this on the server for you (with your permission I could clean this up on the backend which would then propagate to your client(s) next time you sync)

PeterZaal, 2017-04-10:
Small update:
1) When I save into a local xml file and edit and remove the strange characters from the extradata value with key="Deco model info", after that I am able to reload the xml file. I hink the problem was most likely in the one of these extradata value which was split onto 2 lines like this:
  <extradata key='Deco model info' value='VPM-B/GFS +2 %85à²õ
xþ???' />
2) I uninstalled and reinstalled Subsurface and reimported some dives from the Petrel 2 in a new logbook. There are still strange characters in the extradata following the %85, e.g. : 
<extradata key='Deco model info' value='VPM-B/GFS +2 %85À???â±??' />
  <extradata key='Deco model info' value='VPM-B/GFS +2 %85ˆ??€â±??' />
  <extradata key='Deco model info' value='VPM-B/GFS +2 %850' />
The strange thing is that this is version 4.6.3, and the % is still in front of the GFS number (85). After that there are the extra characters (which sometimes even seem to result in carriage returns, and invalidate the xml)
3) I tried to create a dumpfile, but as soons as I check the 'Save libdivecomputer dumpfile' I always get a 'Dive data import error'. So at this moment I am not able to create a dumpfile :(

glance-, 2017-04-11:
Looking around in my digital scrappiles i found the following xml:
```xml
  <extradata key='Deco model' value='VPM-B/GFS' />
  <extradata key='Deco model info' value='VPM-B/GFS +1 %9' />
```
With which code this was downloaded unclear, but it has the % in the wrong place and the 9 is obviously bogus. Should probably have bin a 90. Could have bin buggy code I was playing around with back then.

Could it be so funny that the % in the extra string works as a format string injection against membuffer.c 's show_utf8/put_format ?

If you download from your Petrel again, into a clean logbook, Look at the values in the "Extra data" tab and check if the "crap" is there before saving? If it looks sane there, the corruption might be that we format-string-attack our self =/

PeterZaal, 2017-04-11:
Hi, yes I reimported some dives from the Petrel 2 into a new logbook, and there are still some strange characters (and the % is before the number), see my previous message above at (2).

glance-, 2017-04-11:
Even before you have saved the new logbook?

glance-, 2017-04-11:
What I'm trying to figure out, is if the crap data there is in the source data we get from libdivecomputer, or if it gets added when we save the data.

PeterZaal, 2017-04-11:
Ah right, I did notice that the crap data was shown on the screen, but I don't remember if I saved it before that, or not. I will have to reimport again without saving and let you know!

PeterZaal, 2017-04-11:
Just reimported some dives from the Petrel 2 into a new logbook. Without saving I looked directly at the Extra data, and the crap data is already there.

PeterZaal, 2017-04-11:
Ok, just a long shot: I am a developer myself, but C has been a very very long time ago for me, so maybe this is just nothing, but the line you mentioned earlier (line 412 in shearwater_predator_parser.c) writes the string into buf. The string is/should be "VPM-B/GFS +2 85%". This is 16 characters. buf is max BUFLEN, which is 16, but that includes the \0. The snprintf function *should* terminate the string at max BUFLEN-1 and add the \0. In line 26-28 snprintf actually uses _snprintf (I assume the _MSC_VER is defined since I am using the Windows version). I found postings that the (older) _snprintf actually does *not* add the \0 string if len >= bufsize (https://msdn.microsoft.com/en-us/library/2ts7cx93(v=vs.110).aspx).
Is a \0 is not appended by _snprintf, the string will overflow until somewhere in memory a \0 is found, adding extra crap data. Could this be the case?
Note this also explain why your "VPM-B/GFS +1 %9" is 9 and not 90, since the max string length is 15.

glance-, 2017-04-11:
That sounds exactly like the case. Good work.

janmulder, 2017-04-12:
@dirkhh: nor sure this is the right way (pull request for our libdivecomputer fork), so please process how you see fit. In addition, it might be worth considering to build the Microsoft distribution of Subsurface to a  newer library set.

dirkhh, 2017-04-12:
Given that I cross compile the Windows binaries... I wonder why they would link against old libraries. Aren't the binaries getting the standard C functions straight from the system libraries? Odd.

PeterZaal, 2017-04-12:
Note that line 26-28 makes it using the _snprintf function instead of snprintf.
_snprintf does not append the terminating null if the formatted string length is equal to or greater than the given count (= BUFLEN). Ref: https://msdn.microsoft.com/en-us/library/2ts7cx93.aspx
The other difference is in the returned output value, but that is not used anyway.
So maybe it's not the old library, but the old function.

dirkhh, 2017-04-13:
I created new "daily builds" with the updated library. Would be cool if you could test this, @PeterZaal 

http://subsurface-divelog.org/downloads/daily/subsurface-4.6.3-222-gfd40a29ec858.exe

A fresh download from your dive computer with this should not create those odd strings.

PeterZaal, 2017-04-13:
Cool thanks @dirkhh I will do tonight and let you know!

PeterZaal, 2017-04-13:
Just did a small test, and reimported some dives into a new logbook, but the junk data is still there !? (and the % is still in front of the number).
Is the change that @janmulder made (changing BUFLEN from 16 to 32) included in this build?

dirkhh, 2017-04-13:
Hmm - what does it say on the About page? Just to make sure that you are running that version.

It 4.6.3-225 should include the fix from @janmulder 

(but it's always possible that I fat-fingered something)

janmulder, 2017-04-13:
Which libdc repo did you use for the build? The github one? or the old not maitained one?

PeterZaal, 2017-04-13:
Yeah I also checked the About page, because I wanted to be sure I really installed the correct version, and it says Subsurface 4.6.3-225-g5f9ca5f2ecab

dirkhh, 2017-04-14:
:man_facepalming: 
Yes, indeed, I fat-fingered it. I had updated the sources alright - but I forgot the magic "touch build.libdivecomputer" that is required in my Windows cross-build environment - so this was linked against the old, precompiled libdivecomputer binaries.
A new build is up, this time I verified that it uses the latest libdivecomputer with Jan's changes.

http://subsurface-divelog.org/downloads/daily/subsurface-4.6.3-230-gf8203b8efe47.exe

PeterZaal, 2017-04-14:
Really weird, but I am sorry to say that the junk data is still there :( (yes it's the 230 version).

PeterZaal, 2017-04-14:
But in the libdivecomputer-0.dll I found this string: "VPM-B/GFS +%u %%%u"
This is the old string, it should be "VPM-B/GFS +%u %u%%", so it seems somehow there are still old sources being used.

dirkhh, 2017-04-14:
/me walks outside into the rain, crying silently.
I hate Windows. I hate Windows. I hate Windows.
OK, back to the drawing board of understanding why the correct library is in the staging directory from where I package the installer, yet the wrong library appears to be bundled in the installer.

dirkhh, 2017-04-14:
OK, famous last words. Please try this one. I unpacked the installer and checked the libdivecomputer in there. And it's the right one. Seriously. This time.
http://subsurface-divelog.org/downloads/daily/subsurface-4.6.3-235-g2011f88befee.exe

PeterZaal, 2017-04-14:
@dirkhh It works! No more junk data, and the % is after the GFS number, so definitly the most recent version now! Thanks a lot!

janmulder, 2017-04-14:
@dirkhh: just curious. How do you cross build? I setup mxe today and with the mxe build script it was relatively painless to get my first windows cross build.

janmulder, 2017-04-14:
fixed in https://github.com/Subsurface-divelog/libdc/commit/84eb59c3ce5fbdb2a34e717079ed46f38ea516e7

closed 2017-04-14 by janmulder

dirkhh, 2017-04-14:
That's exactly how I do it. That's why we have that script :-)
The script (for performance reasons) doesn't rebuild all the components. And mine here tries hard to  verify that no malicious code somehow makes it into the build. And an unintended side effect it ended up grabbing the wrong binary to package :-(
Anyway, glad this is finally fixed on the fifth attempt

neolit123, 2017-04-14:
@janmulder, @PeterZaal 

for this use case BUFLEN 32 should suffice.

snprintf() is a not the most portable of functions. MSVCRT or even older glibc are guilty of not following the standard about this one.

unless i'm mistaken, that _snprintf() is not used because we don't define _MSC_VER, which is a MSVC macro and we are using mingw instead. but it doesn't matter because the only function that always null terminates in mingw is __mingw_snprintf(), which is a custom implementation.

the only way to make sure this works on all systems is to either provide a large enough buffer or null terminate manually at `BUFLEN-1`.

janmulder, 2017-04-14:
Fully agree @neolit123. However, as the code is very specific for the Shearwater computers, and the length of the string to be generated is explicitly defined (so the max length is know upfront), the new length will do, as long as no developer get more verbose in the reporting of this deco data.

neolit123, 2017-04-14:
@janmulder 
that's what i said:
> for this use case BUFLEN 32 should suffice.