subsurface/subsurface #308: Integrate wit mention-bot

glance-, 2017-04-15:
https://github.com/home-assistant/home-assistant/ uses mention-bot and I tink it works quite well. It looks at a PR and the history for that particular code thats changed, and mentions those persons in the PR, as potential reviewers.

I think its a great way to involve more people in the review-process.

https://github.com/facebook/mention-bot

dirkhh, 2017-04-15:
That does look cool. I'm a bit strapped for time, trying to get Subsurface-mobile to the point where we can make another release... would you dig into this some more and give me pointers how to set this up?

glance-, 2017-04-15:
I think https://github.com/facebook/mention-bot#how-to-use is pretty clear.

glance-, 2017-04-17:
https://github.com/Subsurface-divelog/subsurface/pull/322 <- YEY - looks like it würks!

closed 2017-04-17 by glance-

atdotde, 2017-04-20:
Cannot find the .mention-bot configuration file. Where did you configure it? I would like to add myself to me mentioned for all commits touching *plan*.c* files as well as deco.c

janmulder, 2017-04-20:
Dirk just activated it, without configuring any specials. So there is no config file yet.