subsurface/subsurface #311: Subsurface-mobile: delete from dive list, missing undo button

dirkhh, 2017-04-15:


neolit123, 2017-09-23:
@janmulder 
might be resolved already.

neolit123, 2017-11-17:
tested the mobile-app on desktop yesterday and the button seems present.

janmulder, 2017-11-17:
@neolit123. I think you are confused. There is a button (long press on dive list) to delete, but no undo capability such as in the delete from dive details. So, re-opening it. 

And very tempted to fix this simple one ... but ... there is something weird lurking around here (as well). Delete works perfectly, no code change, compile again, delete segfaults. So do not dare to touch things here :-)

dirkhh, 2017-11-17:
My suspicion is that we somehow get the model out of sync. We delete the dive from the dive list and from the C++ model. And then accessing that model from QML crashes. Sometimes. So is it possible that QML tries to access an outdated version of that model and depending on what actually happened in memory this may work or may crash? I have tried to track this down for so long :-(

janmulder, 2017-12-14:
Small update. While working on dd87350cb7b3ce7d, I got the idea that the strategy used there might be useful here (getting rid of the crash, that is). In fact it is.  I tried to factor out the beginRemoveRows/endRemoveRows by just removing it from the underlying data, and emitting a dataChanged event. And indeed, that does not/never crash. But, I cannot get the list repainted correctly... so, a little more insight, but no solution (yet).

janmulder, 2018-01-08:
Fix added for the crash ... but failed to implement the undo as asked for here.

Cloning the code from the delete from dive details (which is working perfectly fine, also without the change of #1029) is not straightforward, and I stopped working on it, when I hit another crash on selecting undo on the passive notification (that would probably mean another PITA debug and hacking session).

This all said. Delete from divelist is working with the mentioned patch. Why do we need an undo here? First you need to do a long select, then hit the dustbin within a couple of seconds. That is not something you do by accident. Notice that this is very different on the details page delete. As the delete is part of the action button, you might miss-hit and accidentally press the delete. Besides this, as long as the local cloud storage is not saved instantly at every modification to the online storage, it is even simple for a user to cancel an accidentally deleted dive by closing the app. And the experienced git user just does a git reset --hard on his repo to undo things :-)

So, I'm more or less proposing to close this issue, and merge #1029.

janiversen, 2018-06-22:
I too am in favour of closing this issue, it is impossible to loose a dive by deleting it. If we wanted to 200% solution, we would add a screen, that presented the git revisions (like a trash can) and allow a user to select an old version (however I think we have many other more important issues with mobile).

dirkhh, 2018-06-22:
I agree. The delete on the details could reasonably easily be tapped by mistake. So the undo seems important. It takes quite a bit of effort to get to this second delete from the dive list. Let's just declare that part a feature :-)

closed 2018-06-22 by dirkhh