subsurface/subsurface #313: Subsurface-mobile: GPS upload hang and errors

dirkhh, 2017-04-15:
reported by @willemferguson 

INTERNAL SERVER ERRORs in the app log

dirkhh, 2017-04-15:
I cannot reproduce this and I tried from several devices.
If you can reproduce this and tell me the time you did this and the IP you most likely connected from, I could try to look at the server logs to see if there's anything suspicious there...

Ghalr, 2017-04-23:
I just tried to use for the first time "set current location", I was not sure if it worked or not because it took really long time. Finally it worked and gps position is accurate, but using a high precision option (bluetooth, wifi, gps) /without Glonass I feel it was long for the first time.

Subsurface-mobile version : Subsurface-mobile-4.6.3.274-arm.apk 2017-04-21 11:52
from https://subsurface-divelog.org/downloads/daily/?C=M;O=D

I also found hard to read GPS coordinate but I don't remember if it is correctly written or not:

xx°xx'xx.xxx"S xx°xx'xx.xxx"E

Cannot be easier to read ?

neolit123, 2017-09-23:
@willemferguson @dirkhh 
closing due to lack of user feedback.

closed 2017-09-23 by neolit123