subsurface/subsurface #314: Subsurface-mobile: improvement idea: bundle user manual

dirkhh, 2017-04-15:
suggested by @willemferguson 

This way you can access the user manual when you don't have connectivity

jbygdell, 2017-05-10:
This one was a bummer, the webkit module that gets shipped with android can't read local data so there is no way of bundling the html manual and displaying it.
The other option is to add the manual as a kirigami page, maybe even split it over multiple pages for the different chapters.

dirkhh, 2017-05-10:
Is there a way to automate the conversion into kirigami pages?

glance-, 2017-05-10:
I think we can use http://doc.qt.io/qt-5/qtwebview-index.html and inject the content into the webview using something like https://developer.android.com/reference/android/webkit/WebView.html#loadData(java.lang.String, java.lang.String, java.lang.String)

That way we can read the resource ourself, and just set the content of a webview to it.

neolit123, 2017-09-23:
@glance- do you think this is still doable or perhaps we should move away from the idea?

glance-, 2017-09-24:
I think its doable, and probably even the preferred way of doing this, if we're going to bundle the manual.