subsurface/subsurface #315: Subsurface freezes when text entered at very beginning of dive notes textbox

sfuchs79, 2017-04-15:
I have a freeze of Subsurface in the following setup and situation:
- Windows MXE (Qt 5.7) build (release and debug), current git
- Start Subsurface
- Open new logbook
- Plan a dive in the planner
- Save this dive
- Try to put the cursor at the very beginning (before first caracter) of the text in the dive notes field to edit the notes
--> Freeze in ~1/3 of the times, maybe dependent of the other environment (speed of PC,...)

See the call stack attached below, this may be the start of the interesting part:
0041DF74 4F917810 03C7CC84 00000000  subsurface.exe!fromCursorTilWhitespace  [/home/stefan/Entwicklung/Subsurface/subsurface/desktop-widgets/simplewidgets.cpp @ 761]
0041DCAA 4F917810 03C7CD10 03C7CCE8  subsurface.exe!tryToFormulateUrl  [/home/stefan/Entwicklung/Subsurface/subsurface/desktop-widgets/simplewidgets.cpp @ 734]
0041D53F 4F793940 03C7D3CC 03C7CD88  subsurface.exe!eventFilter  [/home/stefan/Entwicklung/Subsurface/subsurface/desktop-widgets/simplewidgets.cpp @ 664]

[drmingw_subsurface_enter_notes_crash.txt](https://github.com/Subsurface-divelog/subsurface/files/923853/drmingw_subsurface_enter_notes_crash.txt)

sfuchs79, 2017-04-16:
One more comment here:
I for sure also try to understand what this "TextHyperlinkEventFilter" thing is supposed to do and why it could be broken. I understand what it theoretically and basically should do. Detailed explanations wellcome. But I can't see it working in real life. Does it actually work today and where (in the dive notes?!) and how?

sfuchs79, 2017-04-16:
@pestophagous 
Now learned even more. Yes this works if I press Ctrl and click on a link. That brings me to the question how this could be even triggered in my error case because I for sure don't press Ctrl in this case.

sfuchs79, 2017-04-16:
One step forward:
Im now 99% sure that the TextHyperlinkEventFilter is triggered unintentionally for a QEvent::ToolTip and causes the freeze in my case. For the QEvent::ToolTip I'm really not sure what this is good for. Do we want to be able to click on a URL in a ToolTip? If yes where?
https://github.com/Subsurface-divelog/subsurface/blob/master/desktop-widgets/simplewidgets.cpp#L645
Maybe we can remove this path. I tried and it removes my crashes but still keeps the function for enabling the CtrlClick on a URL in the notes.

dirkhh, 2017-04-16:
I don't know why we have the tooltip handling there. This was added by @pestophagous a year and a half ago...

sfuchs79, 2017-04-17:
After reading the code more carefully:
The idea is to show a tooltip if there is a valid URL below the cursor in the dive notes which could be clicked with Ctrl-click. This in general works in current git and I for sure don't want to remove it.
So I'm back at figuring out why sometimes in fromCursorTilWhitespace this crashes in my case:
https://github.com/Subsurface-divelog/subsurface/blob/master/desktop-widgets/simplewidgets.cpp#L761

PR #323 (2017-04-19): Bugfix handle url tooltip crash
This fixes #315  
hopefully...
with no side effects...
finally....
closed 2017-04-19 by dirkhh