subsurface/subsurface #332: At end of DC download (HW OSTC 2) a string is printed in progress bar "xx%" field

sfuchs79, 2017-04-21:
A strange thing that happens at the end of the DC download:
At the position where there has been the "xx%" value right of the progress bar a string related to the last downloaded dive with date and time is printed. Please see the attached screenshot.
[image]

This happens in current 2.6.3-260 Windows daily build from Dirk and also in my own MXE Windows builds. Any release build seems to continue to work properly after this happens. Any of my own MXE based debug builds do also crash after this happens. For sure I don't know exactly if the appearence of the string and the crash are really linked.
For the crash see:
[subsurface_download_crash_qt5.7.1_MXE_2016-12_debug_build-1.txt](https://github.com/Subsurface-divelog/subsurface/files/947296/subsurface_download_crash_qt5.7.1_MXE_2016-12_debug_build-1.txt)

Neither the strange string nor the crash do appear with a linux build.

sfuchs79, 2017-04-21:
One comment: I already identified the most likely source of this string simply based on the content. It seems to be created here:
https://github.com/Subsurface-divelog/subsurface/blob/f8203b8efe47120de73b35643ac56809b44275a5/core/libdivecomputer.c#L613

glance-, 2017-04-21:
If libdivecomputer can provide any better information to us than just the progress, we show it there, so this will differ between different libdivecomputer backends.

Have you tried running your builds in english locale? Just trying to eliminate the whole translation bits.

sfuchs79, 2017-04-21:
The above screenshot is already from a test where I set to english locale.

dirkhh, 2017-04-21:
Different things are shown in this dialog, depending on what you UI toolkit supports. So KDE, Gnonme, Windows and Mac all show this information differently. It's infuriating.
The crash happens in onDownloadThreadFinished() when it calls setImportedDivesIndexes which calls beginRemoveRows(QModelIndex(), 0, lastIndex - firstIndex);
I'd insert some debug output right before that to figure out why that would crash.

The reason this doesn't crash in release builds is because there the asserts are disabled :-)

sfuchs79, 2017-04-21:
Ok you are right, the string printed next to the progress bar and the crash are completely unrelated.
I disabled printing the alternative text to the progress bar and the crash is still there.
The content of the string printed in my case is imho completely useless. Also it doesn't update in any way and stays there even if you repeat the download instead of showing the progress in "xx%" again. Does it really show s.th. more nice and useful elsewhere?

Asserts?! Well yes, I still have to learn so much...
What can I look at beside the values of last and first?

dirkhh, 2017-04-21:
The asserts should make sure that things crash when  you get into a "this should never happen" situation. I'm not a huge fan, because crashing isn't really a useful behavior, but whatever. Knowing first and last might help figure out what's wrong. At this point I'm stabbing in the dark.

As for useful texts - yes, normally this should show progress, i.e. the current dive being downloaded and dive time/date. That used to work with the OSTC backend (but I no longer own a working OSTC, so I can't test).

sfuchs79, 2017-04-21:
Ok, I get the following when I try to import from my OSTC into an empty log:
setImportedDivesIndexes first, last:  0 ,  31
setImportedDivesIndexes firstIndex, lastIndex:  0 ,  -1
ASSERT: "last >= first" in file itemmodels/qabstractitemmodel.cpp, line 2695

Can it be that the only thing that is wrong here is the assert itself???

sfuchs79, 2017-04-21:
Ups, no. The assert that triggers is a completely different...

dirkhh, 2017-04-21:
Ah, fudge. That Q_ASSERT should of course be followed by

if (lastIndex >= firstIndex) {
	beginRemoveRows(QModelIndex(), 0, lastIndex - firstIndex);
	endRemoveRows();
}
Can you add that and try again?

sfuchs79, 2017-04-21:
Yep, that would have been my guess now as well. I'll try. Give me some hours... still have no Ryz**.

sfuchs79, 2017-04-21:
That was just a joke whining again about my poor HW :-) I can tell you building MXE itself is the hell!
Rebuild marble as well because I also still have this odd issue with marble logging output...

Yes, this thing now works. Thanks for the help! PR will follow.

closed 2017-04-21 by sfuchs79