subsurface/subsurface #335: Info text printed into progress bar during DC download (OSTC 2N) not really useful

sfuchs79, 2017-04-21:
During DC download there is some text coming from libdc printed next to the progress bar.
In my example with a HW OSTC 2N wie USB this text replaces the "xx%" value only at the end of the download which is not really usefull or nice.

If anyone is able to fix this in a way that while downloading we have constant updates of the text showing e.g. the current dive beeing downloaded that would be nice. Otherwise I personally would remove this feature and only&always show the "xx%" value.
If this already today gives much nicer results for other DC then keep it as it is.

See picture:
[image]

dirkhh, 2017-04-22:
Looking at the code, this text appears to come from libdc_header_parser() in libdivecomputer.c and should be updated for every dive that you download. Are you saying that even if you are downloading multiple dives (i.e., downloading into an empty logbook), the text only shows up once, at the very end, with one text talking about a "Dive 1"?

sfuchs79, 2017-04-22:
It shows up only at the very end of the download with one text talking about the **last dive** downloaded. So if there is only one dive it says "Dive 1:..." if n dives "Dive n:...". All the other time before I see the "xx%" value beeing displayed correctly.

dirkhh, 2017-04-22:
I played a bit, using the libdivecomputer simulator. And indeed, for those dive computers that parse while downloading, the updates are useful. But for divecomputers like the OSTC that first download, then parse, it's not.
Of course now I need to figure out how to tell a from b - but I think I have an idea.

closed 2017-04-22 by dirkhh

janmulder, 2017-04-23:
@dirkhh: the libdivecomputer simulator ... where can I find that? I found (an old) src tarball on Jef's website, and precompiled binaries, but I would like to use it for a recent libdb version, not some old (end 2014 libdc code). 

I am trying to find a CCR setpoint related bug (which looks suspiciously related to the the TestParse issue you put on the mailing list), but up to now, I am not sure if it is OSTC3 firmware, libdc or ssrf code. So with this, a simulator would be nice.

sfuchs79, 2017-04-23:
@dirkhh 
Works here as expected. No more useless text at the end, only the xx% values. Thanks!

Unfortunately I now saw another not translated text right there. I will change it and do a pull request. Don't know if you want to do the translation loop again. The text is for sure minor important.

glance-, 2017-04-23:
@janmulder the libdivecomputer simulator is a series of patches ontop of libdivecomputer, and the simulator doesn't have any support for simulating a OSTC3. I have some skeleton code for that laying around but nothing thats even close of working.

commit 82d37e74 (2017-04-22): Download from divecomputer: don't show bogus progress text

On some divecomputers we download all the data from the device and
then parse the data afterwards just in memory (at which point the
progress bar has already run all the way to 100%). So don't try to
show the dive number and time in that case.

Fixes #335