subsurface/subsurface #337: [Idea] Gas consumption alert for security reserve and 1/3 rules

Ghalr, 2017-04-22:
Hello. 

What do you think about some visual alerts (warning colored icon) :
- security reserve = red
- 1/3 gas rule = orange

This will automatically display a "warning colored icon" in case we reach a certain gas pressure during the dive. 

Regards,

dirkhh, 2017-04-23:
That's doable on the desktop version... on the mobile version I think that would add too much visual clutter (smaller display in general). But adding such warnings as options... sure.
We already visualize changes in SAC rate. If you suddenly use 60l/min your air consumption line will be bright red.

[image]

For example, in this picture you see a dive where one of the divers had problems equalizing, the rest of us where waiting for him to descend and didn't realize that the current pushed us to the edge of the reef. And then we had to swim back against the current to be able to have an interesting dive. SAC rate when way up as we swam back...

Ghalr, 2017-04-23:
In the Subsurface-mobile indeed it will add much visual effects on it.

Hopefully I have not reach any reserve pressure or experienced a OoG situation, but how are they display actually ? 'cause having automatically a warning icon will be interesting. If we reach 1/3 rules or any calculated gas value for ascent, both 1/3rule / reserve are still important.

mikeller, 2026-10-18:
Hi @Ghalr,

dirkhh was open to adding gas-reserve/rule-of-thirds warnings as a desktop-only configurable option (explicitly not for mobile, due to screen-space/clutter concerns there), building on the existing SAC-rate-based pressure-line colouring rather than duplicating it. To help scope this properly: what's your preferred exact reserve/thirds rule (e.g. 1:3, 1:2, a configurable ratio) that the warning should check against? That'll help us design the actual threshold logic.