subsurface/subsurface #338: [Idea] Subsurface-mobile: to Add a dive; set rules for the form

Ghalr, 2017-04-23:
Hello.

I wonder if it would not be great to have some rules for the form (Add a new dive), according to our metric / imperial units setup, this is more user friendly. For depth when we add 80, according to our units it will display automatically 80m or 80ft

- Depth : only numbers
- Air/Water temp : only numbers
- Time : only number  x:xx (H:mm) 0:40 for 40min for example
- Weight : only numbers
- Initial/Final Pressure : only numbers

Well I guess you see what I mean

dirkhh, 2017-04-23:
The reason why we allow the user to explicitly enter the unit is this:
You are a recreational diver, didn't bring your dive computer on the trip and want to manually enter the drive. You are from the US (so your software is set up for feet/psi/F) but you are diving in Japan, so the dive master tells you "the water was 25 degrees, we went down to 19 meters". No you can enter just that "25C" and "19m" and Subsurface-mobile will do the right thing.

On the other hand, if you just enter "19" then the software already assumes that you mean the "currently active units", so "19" would mean "19ft".

So I think we are doing what you are asking, we only do more / smarter / better.

What do you think? :-)

Ghalr, 2017-04-23:
I understand better, why not directly do something like this, and indeed we actually no need to worry about a unit settings.

100ft              press enter, this will auto-display
100ft / 30m   double conversion

60m               press enter, this will auto-display
60m / 200ft   double conversion

Will be the same for pressure, temperature.

dirkhh, 2017-04-24:
Programmatically that's not trivial to do - when do you add the additional information? Remember, this is a touch screen. People may not press "enter", they might just hide the keyboard. There is no simple consistent concept of "user left this input field".

Another problem would be that it wouldn't show up like this. It would be 100ft / 30.4m or 30m / 99ft
And on smaller displays, this would likely make the screen very crowded.

And of course, once the user saves, they are already seeing the value in their preferred units.

Let's just say you haven't convinced me, yet, that this is an improvement of the user experience.

Ghalr, 2017-04-24:
I see, I'll come back later and edit with more argumentations. 
(I'm not used to the Android version yet.)

Ghalr, 2017-04-27:
@dirkhh 

Hi

I have few minutes to write down my point of view and try to convince you or at least share my feedback about it. After thinking, I admit I had no idea why I suggested exactly this idea, mostly because I would personaly enjoy an app telling me to input numbers only into a number field (Depth) (and telling me I made something wrong) and have double convert input. This is more easy for divers to directly read their respective units.

I believe divers are well-organized and it will not be a problem for them to input numbers only into the specific field, and to answer your question I might only use the app to transfert/sync latest dive into the cloud and share after I finish the dive. Or I am using a paper and then do add the dive when I have time. So I'm adding additionnal infos when I gathered all infos about it (example visibility, comments etc etc) - and have converted units.

I think it is more userfriendly to plan / add new dive using a 10' tablet than a 5' smartphone; some tablet can have a SIM card inside for 3/4G network so this is more fine to use Subsurface mobile app. 

I have not yet experienced a dive where instructors/buddies used the imperial system but as I already said somewhere here, it will not be a problem for me to know most used units conversion from my depth limitations (I printed a conversion table ready to use). Also displaying a double conversion units can be really great to share easily with divers who use imperial system, there are a sort of respect where I would enjoy that a foreign diver speak to me into a metric system as I am using it. And I am pretty sure it will be the same in return.

The result can be taken at the upper round anyway, exemple :
3m - 10ft
6m - 20ft
9m - 30ft
15m - 50ft
30m - 100ft
40m - 130ft

PADI imperial table got those detph values : 35 40 50 60 70 80 90 100 110 120 130 140
PADI metric table got those depth values :   10 12 14 16 18  20 22 25   30   35   40   42
35ft = 10,668m
100ft = 30,48m
140ft = 42,672m

In any case we do use the diveplaner or the dive table with common units.

Might I finally suggest 2 things afterall:
- depending of our current unit system settings
on each "add new dive" form have :
input unit system :   O Imperial        O Metric
so divers will put directly number into the correct in use unit system

- Display final double most common conversion (upper round) after saving
30m/100ft  if metric system
100ft/30m  if imperial system

In any case after few times playing about lenght system we already learn by heart some values.

dirkhh, 2017-04-27:
I think you make reasonable arguments. But based on feedback from our users over the past few years, here are a few caveats:

a) in general, divers HATE it when the software rounds depth values. We did this and people were not happy.
b) while there are people who crave as much information as possible, most people crave easy and obvious. So when you use Subsurface and Subsurface-mobile, they try to automagically give you the units you are most likely to want (based on the locale in which you run your computer / tablet / phone)
c) the concept of least surprise is very important when creating a UI. So when a user enters only numbers, they will be least surprised to get those interpreted in the units that are being displayed. Conversely, if they want to enter the values in different units, the logical thing to do is to just type in the units.

Fundamentally, I don't see how what you are suggesting actually makes things better for the majority of our users. As you say, you have never met someone who used imperial units. And that is true for most divers in Europe who don't travel to the Caribbean or the US. So why should we clutter the display with Imperial units? On the flip side, most US divers diving in Hawaii or the Caribbean have no idea what they should do with metric information, so why clutter their display?

I appreciate all the suggestions that you make - and quite a few of them have helped us make Subsurface and Subsurface-mobile better. But this one I'll close as "won't do".

closed 2017-04-27 by dirkhh

Ghalr, 2017-04-27:
I do understand at least I tried to share my point of view.
I was suggested by my OW instructor to learn to use/understand both units system to be able to clearly have some benefits from it (since the beginning), cause you do not know if you got opportunity to travel in Florida, met some US instructor and speaking in metric system would certainly not help much.

- most diving articles/scuba forums/YT video are written in English, and mainly using imperial system
- be confident using both units system
- prevent misunderstanding about depth value
- report back divelog using both imperial/metric system will improve visibility in case you write down a blog or something to share your dive trips.
- etc etc

But I'm agree US divers will not use metric system, so clutter their display is not interesting.