subsurface/subsurface #339: [Question/Bug]? Strange to use Portrait mode, no Zoom-out available ?

Ghalr, 2017-04-23:
Hello.

Using a 5" pouce smartphone I feel it is strange to use Portrait mode as even in english or french each words exceed the width, and "Add new dive" form is not clearly usable into Portrait mode.

Would it be possible to use Subsurface-mobile only in Landscape mode (set as default), because this can make run-away new user.

I am not able to use a Zoom-In / Zoom-Out (with 2 fingers) to be able to have small font, and better display "Add new dive" form. Is it intended to work without Zoom feature ?

Best regards,

dirkhh, 2017-04-24:
No zoom is one of the frustrating limitations of the ui toolkit that we decided to use.

I don't want to lock this to landscape only - I think what would be a better solution, given the limitations mentioned above, would be an option to switch to a smaller or bigger font.

Would that address your problem?

I'll have to play with this to see how that could be implemented in a seamless fashion, but once 4.6.4 is out that is something I could look into.
(maybe you could adjust the title if you agree with this approach - just so that the issue / label reflects what we actually would try to add...)

Ghalr, 2017-04-24:
I understand.

Yes this will really help to have small font (as default), I used a 10' tablet and this is fine, font are just too big for 5' screen size.

I will try to check if a screenshot is possible on Android and I come back later, for sure having small font will really help for user who prefer Portrait mode.

neolit123, 2017-09-23:
@janmulder not sure if this was added already.

janmulder, 2017-11-16:
I propose to close this issue. Layout has changed a lot since April and current 2.0 version work just fine on my 5.5" device in portrait mode with respect to font size.

closed 2017-11-16 by neolit123