subsurface/subsurface #342: OSTC3 incorrect notice to upgrade firmware

janmulder, 2017-04-23:
Pulled a libdc dump from my OSTC3. At the end of the download of the dump, the message appears that the firmware needs to be updated. In itself nothing special, but it tells: you have version 0.0, and latest stable is 2.15 (at time of writing this issue). The 0.0 is wrong, as the OSTC3 has 2.15 installed.

Definitely triggered by the selection to make a libdc dump, as the normal download works just fine. So definitely nothing urgent.

glance-, 2017-04-23:
The reason is that we never get a `DC_EVENT_DEVINFO` when doing a memory dump.

Something like this should fix it:
```diff --git i/desktop-widgets/downloadfromdivecomputer.cpp w/desktop-widgets/downloadfromdivecomputer.cpp
index 4a636a7c..4b2832cf 100644
--- i/desktop-widgets/downloadfromdivecomputer.cpp
+++ w/desktop-widgets/downloadfromdivecomputer.cpp
@@ -387,10 +387,11 @@ void DownloadFromDCWidget::on_downloadCancelRetryButton_clicked()
        // the user entered in the dropdown.
        // You can enter "OSTC 3" and download just fine from a "OSTC Sport", but
        // this check will compair apples and oranges, firmware wise, then.
-       QString product(ui.product->currentText());
-       if (product == "OSTC 3" || product == "OSTC 3+" ||
+       //
+       // We shouldn't do this for memory dumps.
+       if ((product == "OSTC 3" || product == "OSTC 3+" ||
                        product == "OSTC Cr" || product == "OSTC Sport" ||
-                       product == "OSTC 4")
+                       product == "OSTC 4") && !data.libdc_dump)
                ostcFirmwareCheck = new OstcFirmwareCheck(product);
 }
 
```

dirkhh, 2017-04-24:
This seems reasonable... can I get this as a patch / pull request?

closed 2017-04-24 by janmulder