subsurface/subsurface #345: Subsurface-desktop: Preferences>Georeference>Dive Site Layout : Strings are not translated, even if in Transifex

Ghalr, 2017-04-24:
Hello.

Tried French and German language and I noticed that strings into this category (Georeference>Dive Site Layout) are not translated even if they are in Transifex (and translated).

Subsurface-version : subsurface-4.6.3-287-g51ee9d4405d5.exe | 2017-04-23 12:54 | 57M

dirkhh, 2017-04-24:
Hmm. Odd. Looks like a bug. I'll investigate.

sfuchs79, 2017-04-24:
@janmulder 
The easy way of fixing it by only adding the missing tr(); doesn't seem to work. Could this be similar to:
https://github.com/Subsurface-divelog/subsurface/commit/5c89115efe3bf542b3885ce4a1c6dfc03e755b95

janmulder, 2017-04-25:
@sfuchs79. Could well be something like the one I fixed in https://github.com/Subsurface-divelog/subsurface/commit/5c89115efe3bf542b3885ce4a1c6dfc03e755b95.

janmulder, 2017-04-25:
@sfuchs79 (and @dirkhh). I have a fix, but running now time now to do a PR. First thing tomorrow.

dirkhh, 2017-04-25:
Awesome - I won't do a release until later this week, possibly the weekend, so this is fine.
Thanks for the heads up

Ghalr, 2017-04-25:
Thank you so much.

PR #352 (2017-04-26): Correct divesite geocoding translation
See https://github.com/Subsurface-divelog/subsurface/issues/345. The menus where not translated. The basis of this error is a simple typo in core/taxonomy.c where the classname was mis-spelled in the QT_TRANSLATE_NOOP. In addition, to pull and translate the strings from C code, the normal tr() does not work, and the functionality from the gettextfromc class is used.

Fixes: #345
commit 12a59ef3 (2017-04-26): Correct divesite geocoding translation

See https://github.com/Subsurface-divelog/subsurface/issues/345. The
menues where not translated. The basis of this error is a simple
typo in core/taxonomy.c where the classname was mis-spelled in the
QT_TRANSLATE_NOOP. In addition, to pull and translate the strings
from C code, the normal tr() does not work, and the functionality
from the gettextfromc class is used.

Fixes: #345
closed 2017-04-26 by dirkhh