subsurface/subsurface #347: Equipment tab column width not preserved across session

janmulder, 2017-04-24:
The 2 tables (cylinders and weights) on the equipment tab do not preserve changed column widths across sessions. Currently not sure when this got broken.

dirkhh, 2017-04-24:
git bisect is your friend :-)

janmulder, 2017-04-24:
Exactly :-) something for tomorrow.

janmulder, 2017-04-25:
Ok, bisected it to 2f86cbdc3690aa0e5ef8e321f4d5dbb10bf640cc. And the change in tableview.cpp is the main suspect. But revering this results in a SIGSEGV, and in itself the change looks sound ... this all part of @tcanabrava magic rewrite of the monster maintab stuff.

sfuchs79, 2017-04-26:
One interesting finding on my side:
Current master **does** restore ("load") the column widths. But it **doesn't** save them.
Way to test this: Parallel install v4.6.3 release and current master. Changes done while running v4.6.3 are reflected when afterwards running current master. But not vice versa.

janmulder, 2017-04-26:
Yes, I saw this by just looking at the file that holds these sizes. And this is introduced at the mentioned commit, but I really do not understand why that is.

glance-, 2017-04-26:
The with is saved by the destructor `TableView::~TableView` .

It could be c88be99d84aa0de63b77fcaa96c40b73ea10ae8c or 1fc4fba69fed8607d9b961086a88bcf362a482a5 that masks the real problem, which i think is that the Tableview model is already freed and removed, when the destructor is called.

janmulder, 2017-04-26:
https://github.com/Subsurface-divelog/subsurface/commit/c88be99d84aa0de63b77fcaa96c40b73ea10ae8c seems unrelated (that is, it just protects against a crash). And indeed that crash is most likely caused by the already freed model when the tableview is destroyed. So agree with you.

To me it seems pretty sure that https://github.com/Subsurface-divelog/subsurface/commit/1fc4fba69fed8607d9b961086a88bcf362a482a5 introduces that too early free of the model, but as that is a huge commit, it is difficult to understand why (and where) it introduces that.

commit 7aac3852 (2017-04-26): Fix saving table size

We need to make sure the model is avaliable for the table
until it hits the destructor.

Fixes #347
closed 2017-04-26 by dirkhh