subsurface/subsurface #348: Subsurface-desktop: Usersurvey do not have "Subsurface" version strings translated

Ghalr, 2017-04-24:
Hello.

I noticed that "Subsurface" version is not translated, everything is fine but I found it interesting to change to

Subsurface version:

This is just a user friendly feature to fit with other strings who got the ponctuation   :    and this will allow to fully translate into our respective language. 

Thanks.

janmulder, 2017-04-24:
you mean the 4 lines at the bottom of the survey? Like (in Dutch):

Subsurface 4.6.3.289
Besturingssysteem: Linux (Arch Linux)
Centrale verwerkingseenheid: x86_64
Taal: nl-NL

Well, this looks perfectly translated.

You seem to suggest to change the fist line to "Subsurface versie: 4.6.3.289" ... well ... IMHO not very relevant.

Ghalr, 2017-04-24:
Correct I mean those lines.
and yes this is what I wanted to suggest.

dirkhh, 2017-04-27:
I don't think this really makes any difference

closed 2017-04-27 by dirkhh